Xilinx/RapidWright #1: Improving the installation instructions and process

maltanar, 2018-05-01:
A few suggestions to make the RapidWright installation process go more smoothly:
- **mention what is possible with the Quick Start (standalone jar) approach** -- as far as I understand, the Vivado tcl script (rapidwright.tcl) does not get installed this way. So using e.g rapid_compile_ipi is not possible and requires the Full Installation.
- **don't re-download the rapidwright_data and rapidwright_jars if they already exist** -- I had to run the rapidwright-installer.jar several times (because I already had a directory called RapidWright in the same folder, and again because I found out I had to install javafx), and every time it is run it re-downloads the rapidwright_data and rapidwright_jars files. The former one is 1.2 GB and takes time to re-download, especially on slower connections.
- **provide alternative instructinos for OpenJDK** -- I see that the instructions mention Oracle JDK Java 1.8 or better, but some may prefer the open source OpenJDK alternatives. For instance, it looks like RapidWright requires javafx, which is e.g. not provided by openjdk-8-jdk under Ubuntu and requires a separate package installation. It would be nice to have apt-get instructions for easier installation on Ubuntu (I'm testing this myself currently and can report back when/if it works).

clavin-xlnx, 2018-05-02:
Thank you for the feedback, I appreciate your willingness to help us make the install process smoother.  

Regarding the standalone jar install, I have yet to update the documentation, but there is an undocumented option "--unpack_data" that expands the data and tcl folders inside the jar and would have enabled using the rapid_compile_ipi command.  Do you think it would be best to expand this data (if not already done so) when invoking the standalone jar?

Regarding the download of the rapidwright_data.zip and rapidwright_jars.zip, I will add some MD5 hashing checks and will not download if the file is already present.  Good suggestion.

I think this is a good suggestion as OpenJDK is often available on most Linux platforms, but I haven't gotten around to testing it.  If you can help us with the steps/dependencies, I can certainly update the documentation.

maltanar, 2018-05-02:
> Regarding the standalone jar install, I have yet to update the documentation, but there is an undocumented option "--unpack_data" that expands the data and tcl folders inside the jar and would have enabled using the rapid_compile_ipi command. Do you think it would be best to expand this data (if not already done so) automatically when invoking the standalone jar?

Ah, great. If the extracted data doesn't take up too much space (I think a few hundred MB is fine), I think it's a good idea to extract the data/tcl folders right away, so that the user can experiment more without having to go to a full installation.

> I think this is a good suggestion as OpenJDK is often available on most Linux platforms, but I haven't gotten around to testing it. If you can help us with the steps/dependencies, I can certainly update the documentation.

I ended up just installing Oracle JDK in the end, but I'll give a new try with OpenJDK and post my findings here. With Oracle on Ubuntu 16.04, I had a library problem for libpng so I had to use the following to be able to launch the device browser:

wget -q -O /tmp/libpng12.deb http://mirrors.kernel.org/ubuntu/pool/main/libp/libpng/libpng12-0_1.2.54-1ubuntu1_amd64.deb \
  && sudo dpkg -i /tmp/libpng12.deb \
  && rm /tmp/libpng12.deb

And thanks for making RapidWright -- I have an installation now and can't wait to try it out.

timobility, 2018-05-06:
Great to see more open tools for Xilinx devices.

I seem to have the compilation working with Ubuntu 16.04 and OpenJDK.
I am somewhat rusty when it comes to Java so there may be improvements that can be made to the following instructions.

Here's what I did:
sudo apt-get install default-jre
sudo apt-get install default-jdk
sudo apt-get info gradle

It seems openJDK doesn't come with javaFX which is needed. So:
sudo apt-get install openjfx

So this gave me openjdk version "1.8.0_162" and Gradle 2.10.

Then follow the manual installation steps on the RapidWright Getting Started page.
The gradle command seemed happy.

Then copy rapidwright.jar from build/libs to jars.
Then run the following command again:
export CLASSPATH=$RAPIDWRIGHT_PATH:$(echo $RAPIDWRIGHT_PATH/jars/*.jar | tr ' ' ':')

Then these two commands worked:
java -cp $CLASSPATH com.xilinx.rapidwright.device.browser.DeviceBrowser
java -cp $CLASSPATH com.xilinx.rapidwright.device.XDDPrinter xa7a15t xa7a15t.xdd

I haven't tried much more yet.

Hope that helps.

gtri-jdanner, 2018-05-15:
Throwing my Red Hat experience into the ring here: gradle 4.7 doesn't seem to properly support JDK 10. I got this working with java 1.8.0_172. Once that was working, I also needed to install libpng. Enjoying playing around with the tools!

clavin-xlnx, 2019-03-01:
Its taken some time to get around to improving the installer, but I think I have finally addressed most of the issues:

1) JavaFX is no longer a dependency of RapidWright
2) The rapidwright_data.zip and rapidwright_jars.zip files are now MD5 hashed before download to avoid re-downloading after a failed install.
3) `git` and `javac` are both tested up front and the installer is halted prior to continuing with instructions given to the user on how to install git and/or the JDK.  
4) Additional information regarding the libpng12 requirement has been added to error messages in the installer and the [Automatic Install](http://www.rapidwright.io/docs/Automatic_Install.html#automatic-install).

closed 2019-03-01 by clavin-xlnx