Xilinx/RapidWright #2: write_xdd_file is not an available Vivado command

timobility, 2018-05-03:
The XDD file creator appears to call write_xdd_file inside a Vivado TCL shell.
This is the source: https://github.com/Xilinx/RapidWright/blob/master/com/xilinx/rapidwright/device/DeviceFilesCreator.java

Vivado 2018.1 reports this is not an available command.
It is not documented in any public version of the Vivado TCL reference manual.

clavin-xlnx, 2018-05-03:
Indeed, 'write_xdd_file' is not a production command and there are no plans to make it so (sorry).  

However, RapidWright comes with a class that can generate XDD files from the RapidWright device database files called XDDPrinter.  You can achieve similar results by running the XDDPrinter from RapidWright.  Here is how you can generate it (assuming you have RapidWright setup on your CLASSPATH -- see http://www.rapidwright.io/docs/Getting_Started.html for details):

`java com.xilinx.rapidwright.device.XDDPrinter <partname> <output.xdd>`

Ultimately, DeviceFilesCreator is just example code in the case that someone wants to make changes to an architecture and re-generate a device file within RapidWright for experimentation.  Obviously, they won't be able to push any designs into Vivado for custom-made architectures, but allows for 'what if' kinds of exploration efforts.

closed 2018-05-03 by clavin-xlnx

timobility, 2018-05-03:
Thanks for the reply.
There five appear to be any xdd files to print?

timobility, 2018-05-03:
Sorry, I was looking in the github repository.
I see there are some released in the package.