Xilinx/RapidWright #5: SiteType class not visible

mmicko, 2018-10-06:
In class com.xilinx.rapidwright.device.Site there is a method public SiteType getPrimarySiteType() but 
SiteType is not exposed in API, so it's methods are not visible to user instead it return mangled class  com.xilinx.rapidwright.device.c

clavin-xlnx, 2018-10-08:
Thank you for pointing this out.  This method should return a value of the type SiteTypeEnum and will be fixed in the next release.  Any data relevant to site types should be accessible through the Site or SiteInst class.

clavin-xlnx, 2018-10-15:
On further inspection, this specific API will be removed.  In the class com.xilinx.rapidwright.device.Site the method "public SiteTypeEnum getSiteTypeEnum()" will return the SiteTypeEnum.  Also, as a result of this issue, I have added the following APIs:

com.xilinx.rapidwright.device.Site "public SiteTypeEnum[] getAlternateSiteTypeEnums()"
com.xilinx.rapidwright.design.SiteInst "public SiteTypeEnum getPrimarySiteTypeEnum()"
com.xilinx.rapidwright.design.SiteInst "public SiteTypeEnum[] getAlternateSiteTypeEnums()"

These APIs will be available in the next RapidWright release, 2018.2.1.

mmicko, 2018-10-16:
Thanks, but can we as well get BEL[] getElements() in Site class ? Now we can do BEL getBEL(String name), but we need to know all names in that case per SiteTypeEnum.

clavin-xlnx, 2018-10-16:
Yes, it seems like these are missing and I have filed a new issue ( #7 ) to track this request.  Thank you for pointing out the inconsistency. The next release will contain these APIs.

closed 2018-10-16 by clavin-xlnx

mmicko, 2018-10-17:
Thanks for taking care of this.