Xilinx/RapidWright #6: Mac OS qtjambi Issues

nelsobe, 2018-10-13:
It would not run on a mac with the provided qtjambi jar files.  Had to upgrade to both qtjambi-4.6.3.jar and qtjambi-macosx-gcc-4.6.3.jar for it to work.  Was getting library load errors with the distributed versions.

clavin-xlnx, 2018-10-15:
Thank you for the report.  Unfortunately, I don't have a test platform for Mac OS and was blind to this fact.  In the past we were having incompatibility issues with 4.6.3 on Red Hat, hence we used the older version.  Since Vivado is only supported on Linux/Windows, we'll need to stick with 4.5.2_01 for now.

I will remove the Mac OS X jar for Qt Jambi from future distributions.  It is useful information though, to know that it can work with Qt Jambi 4.6.3 for other users wishing to run on the Mac.  Maybe with some help, we could create an FAQ or forum how-to post.  Of course, they will also have to navigate API difference between 4.5.2_01 and 4.6.3 until Red Hat support can be establish.

clavin-xlnx, 2018-10-16:
Please see this forum post for further discussion: https://groups.google.com/forum/#!topic/rapidwright/BBcp9b-zT4w

closed 2018-10-16 by clavin-xlnx