Xilinx/RapidWright #8: Request: Method to create an IOB in a Design based on an existing EDIFCellInst

nelsobe, 2018-10-23:
There is no method to create an IOB corresponding to an existing IBUF or OBUF EDIFCellInst and then place it.

Assume you have an EDIFNetlist of a design and now want to create a Design for it and place all the EDIFCellInst's in it.  You load the netlist and create a Design based on that netlist.  You now have the logical design.  Now you want to place the cells

For EDIFCellInst's which are not IBUF's or OBUF's you can do a Design.createCell(String, EDIFCellInst) followed by a Design.placeCell(Cell, Site, BEL, pinMappings) where the pinMappings are optional.  This allows you to base the creation of the Design's Cell on an existing EDIFCellInst rather than creating a new EDIFCellInst.  
 
There doesn't seem to be a similar way to do this for IBUFs or OBUF's.  Calling Design.createAndPlaceIOB(String, PinType, String, String) will create a new EDIFCellInst, rather than basing it on an existing EDIFCellInst. Design.createIBUF() and Design.createOBUF() have similar problems.  

What is requested is a Design.createAndPlaceIOB(String, EDIFCellInst, String, String) method. This would not create a new EDIFCellInst but would use the provided IBUF or OBUF EDIFCellInst.  Otherwise, the other parameters are the same and it would do what the existing version of the method with this same name does.  

The attached file is the equivalent of Lesson1.java but with using this desired capability.
[Lesson5.java.txt](https://github.com/Xilinx/RapidWright/files/2503946/Lesson5.java.txt)

clavin-xlnx, 2018-10-25:
Looking at Design.createIBUF() and Design.createOBUF(), these method should be removed as they are poorly named (they actually place the IBUF/OBUF as well).  They duplicate the functionality of Design.createAndPlaceIOB().  If a user wants to add an IBUF or OBUF instance to the netlist, this can be done using the existing EDIF-class APIs.  

To provide the requested functionality, I will add an API Design.placeIOB(EDIFCellInst bufInst, String pkgPin, String ioStandard).

clavin-xlnx, 2018-10-29:
I have created a new release [2018.2.3](https://github.com/Xilinx/RapidWright/releases/tag/2018.2.3-beta) with the API updates requested and should resolve this issue.

closed 2018-10-29 by clavin-xlnx