Xilinx/RapidWright #9: Vivado crashes during placement or routing

todd1251, 2018-10-27:
I am regularly seeing Vivado placement or routing crash with the same error (sample logs attached) after using `rapid_compile_ipi` to relocate some blocks. Does not seem to occur when blocks are not relocated. Not really sure what triggers it so I haven't been able to create a self-contained reproduction. Any suggestions on what I might try?

Placer crash: [hs_err_pid69626.log](https://github.com/Xilinx/RapidWright/files/2522312/hs_err_pid69626.log)
Router crash: [hs_err_pid76813.log](https://github.com/Xilinx/RapidWright/files/2522313/hs_err_pid76813.log)

clavin-xlnx, 2018-10-28:
Thank you for posting this issue.  It is very likely the cause is that the blocks are not being placed legally (could be overlapping).  One way to determine if this is the case is to use the HandPlacer to perform a visual inspection/modification of the blocks after they are placed in the flow with `rapid_compile_ipi`.  To do this, you would need to modify the flag `OPEN_HAND_PLACER`in the [BlockStitcher](https://github.com/Xilinx/RapidWright/blob/master/com/xilinx/rapidwright/ipi/BlockStitcher.java#L95) and set it to `true` and recompile.  This will open a GUI after placement of the blocks is performed but before the DCP is written out,  The HandPlacer will provide colored blocks on a tile-based grid showing placement of the blocks.  Green blocks are legally placed, red blocks are illegal and orange is legally placed, but overlapping.  You can also use the mouse to move the blocks to try and discover valid placements.

Obviously, we need to improve the feedback to the user so that we can prevent errors like this from getting to Vivado place and route.  If you are able to provide an example design and/or share your IGF file that would be helpful.  Any additional information would also be quite useful.

todd1251, 2018-10-28:
It seems the hand placer only shows up if not using an IGF file. I did get it to show up and the blocks are all green. If I am able to make an example design that shows the problem I will let you know.

Thanks!

clavin-xlnx, 2018-10-29:
My apologies, I realized that the change to enable the HandPlacer to open was not pushed to the repo.  I have just made another release (2018.2.3) which includes the fix, when the OPEN_HAND_PLACER flag is set, it will work for both modes of the flow.

clavin-xlnx, 2019-07-03:
I believe the material problems underlying this issue have been fixed in recent releases.  There was a bug a while back related to poor pblock creation that could have led to this problem.  If a reproducible test case can be found, please reopen.

closed 2019-07-03 by clavin-xlnx