Xilinx/RapidWright #11: Unfound Device Family Type "VIRTEXUPLUSES1"

leoliu49, 2018-10-29:
Hi,

I'm loading in DCP files for part 'xcku5p-ffvb676-2-e' (```Design.readCheckpoint```) which I don't has anything to do with the issue.

However there seems to be a failure caused during the static class initialization at ```PartNameTools```:
```
Caused by: java.lang.IllegalArgumentException: No enum constant com.xilinx.rapidwright.device.FamilyType.VIRTEXUPLUSES1
	at java.lang.Enum.valueOf(Enum.java:238)
	at com.xilinx.rapidwright.device.FamilyType.valueOf(FamilyType.java:1)
	at com.xilinx.rapidwright.device.PartNameTools.<clinit>(PartNameTools.java:60)
```

A closer look reveals that "VIRTEXUPLUSES1" (which I think used to be there before the rerelease) is now gone from ```FamilyType```:
```
...
    VIRTEXUPLUS,
    VIRTEXUPLUS58G,
    VIRTEXUPLUS58GES1,
    VIRTEXUPLUSHBM,
    VIRTEXUPLUSHBMES1,
...
```
Any suggestions?

clavin-xlnx, 2018-10-29:
Yes, the part you are trying to load is independent of this issue.  You are also correct in that VIRTEXUPLUSES1 is no longer a supported family in Vivado and thus, RapidWright no longer has it included. 

I can't be sure as to the cause of your problem, but it is likely that there is a previous RapidWright release (May/June time frame) that is colliding with the latest install.  Try checking your CLASSPATH and RAPIDWRIGHT_PATH environment variables to make sure they are set correctly and only point to the latest RapidWright installation.  Share more info if you can and I might be able to help more.

Thanks!

leoliu49, 2018-10-29:
Got it - thanks!
I had conflated parts of both releases together.

Thanks,
Leo

closed 2018-10-29 by leoliu49