Xilinx/RapidWright #12: Root Source Folder

jakobwenzel, 2018-11-01:
Hi,

currently, there is no subfolder for source files. Instead, the first package directory "com" is directly inside the project's root directory. In the gradle build, this first package directory is specified as source directory.

Normally, the directories named after packages only start inside the specified source directory. While gradle can build this, other tools like IntelliJ do not expect this directory layout. The project can therefore currently not be compiled or run inside IntelliJ.

I would suggest introducing a src directory and placing the com directory inside it.

Regards,
Jakob

clavin-xlnx, 2018-11-02:
This is a reasonable request.  I will look into making this change at our next major release to coincide with other changes.  I just want to make sure I can fully test the consequences of such a change before making it.

clavin-xlnx, 2019-07-03:
A `src` directory has been inserted between `RapidWright` and the `com` directory to solve this problem in 2019.1.0.  A default IntelliJ project file (folder) has also been included in the repo.

closed 2019-07-03 by clavin-xlnx