Xilinx/RapidWright #14: SitePinInst.getBELPin() produces NullPointerException

jakobwenzel, 2018-11-02:
Hi,

I am experiencing a problem with the method SitePinInst.getBELPin. On an xc7a200t, I have a SiteInst of type RAMB18E1, placed on a Site of type FIFO18E1. For multiple of its SitePinInsts (e.g. RSTRAMARSTRAM), calling getBELPin on them produces a NullPointerException within the method.

The method seems to call Site.getBEL() with the Pin's name, which returns null. Calling SiteInst.getBEL() instead returns a BEL. The Javadoc of these two methods is almost identical. What is their difference?

Regards,
Jakob

clavin-xlnx, 2018-11-02:
I can confirm this is a bug.  The SitePinInst.getBELPin() is improperly handling the fact that the SiteInst is configured with an alternate site type.  I will fix this and examine any other related APIs for similar issues in the next release.

nelsobe, 2018-11-05:
This brings up a question I had regarding alternate site types.  Is there the ability to place a cell onto a site where the site type needs to be changed to an alternate type to accommodate the placement?  This would seem to be a useful capability - it may be that Design.placeCell() does this behind the scenes - not sure.  But, my memory is that one of the work-arounds I tried for Issue #8 was to manually place an IBUF onto an IOB and it was failing because the default type for the IOB site chosen would not accommodate the IBUF.