Xilinx/RapidWright #15: SiteInst.getSiteTypeEnum() returns null

jakobwenzel, 2018-11-06:
Hi!

I have encountered an issue with SiteInst.getSiteTypeEnum(). Still on a xc7a200t, I have a SiteInst placed at ILOGIC_X0Y106. For this SiteInst, getSiteTypeEnum() returns null. Is this supposed to happen?

Also, there is a documentation bug: Javadoc claims the method returns a string.

Regards,
Jakob

clavin-xlnx, 2018-11-06:
Hi Jakob - SiteInst.getSiteTypeEnum() should only return null if the SiteInst is not placed and the type was not specified in a constructor.  So yes, it appears to be a bug.  The work around would be to set the SiteTypeEnum of the SiteInst in a constructor, but this should not be required if the default type is desired.  

One item to note is that if the SiteTypeEnum is not set on the SiteInst, it will default to the default type of the Site onto which it is placed.  If an alternate site type is needed, the SiteTypeEnum should be set in the SiteInst constructor.  

Thank you also for the note regarding the Javadoc, I will update accordingly.

clavin-xlnx, 2018-11-07:
I believe the fix I have will cover the situation, but if you are able to share the DCP, that would be helpful.

clavin-xlnx, 2018-11-16:
I believe I have fixed the issue in the [v2018.2.4](https://github.com/Xilinx/RapidWright/releases/tag/v2018.2.4-beta) release.  I created a design with similar characteristics and the error appears resolved.  If there is still an issue, please re-open or file a new issue accordingly.

closed 2018-11-16 by clavin-xlnx