Xilinx/RapidWright #19: no .igf file found

dsp20, 2019-02-07:
I am getting a message that no .igf file available and when I am running command rapid_compile_ipi. Execution stops with showing messages below. Please help me with this issue
Exception in thread "main" java.lang.NullPointerException
	at com.xilinx.rapidwright.design.MetadataParser.parse(MetadataParser.java:357)
	at com.xilinx.rapidwright.design.Module.<init>(Unknown Source)
	at com.xilinx.rapidwright.ipi.BlockCreator.createBlock(BlockCreator.java:101)
	at com.xilinx.rapidwright.ipi.BlockCreator.createOrRetrieveBlock(BlockCreator.java:504)
	at com.xilinx.rapidwright.ipi.BlockStitcher.main(BlockStitcher.java:523)

clavin-xlnx, 2019-02-07:
Providing an .igf file, although recommended, is optional and will not cause a direct problem.  

The error you are seeing is happening during the parsing of a generated metadata file.  When running rapid_compile_ipi, each IPI block is synthesized, placed and routed out of context.  Additionally, the RapidWright tcl command (see tcl/rapidwright.tcl) generate_metadata is run on the implementation to aid in creating a pre-implemented module.  

From the error message it looks as though it cannot find the SiteInst where a site pin resides which is unexpected.  Would it be possible for you to share the DCP in question?  this would be located in the $(HOME)/blockCache directory.   There will be several hash-name-based subdirectories in your blockCache directory.  You may want to add a print statement to the code to identify which block is causing there error.  For example, you could add the following after line 123 to the file com/xilinx/rapidwright/design/MetadataParser.java:

`System.out.println("Parsing Metadata file: " + metadataFileName);`

To answer your other question, rapid_compile_ipi will finish after the Java thread has terminated.  Unfortunately, it won't show the progress in the Tcl command window.  If it finishes successfully, it will print a message:

`Created Placed DCP at: *_placed.dcp`

This code can be found in com/xilinx/rapidwright/ipi/BlockStitcher.java:764.

clavin-xlnx, 2019-07-03:
Without a test case, it is difficult to resolve this issue.  Closing for now.

closed 2019-07-03 by clavin-xlnx