Xilinx/RapidWright #22: LUT6_2 unsupported on Artix 7?

mithro, 2019-02-26:
When I try and use a `Unisim.LUT6_2` I get the following error;
```
java.lang.RuntimeException: java.lang.RuntimeException: ERROR: This unisim LUT6_2 is not supported on ARTIX7
```

clavin-xlnx, 2019-02-26:
It looks like this errors is coming from Design.createAndPlaceCell() and its variants.  I've dug into the cause and it turns out that the LUT6_2 is actually not a primitive in Series 7/UltraScale/UltraScale+.  It is "translated prim" which gets decomposed into fundamental primitives (LUT5 and LUT6, for example).  When you open a DCP, you'll notice this transformation taking place:

    INFO: [Project 1-111] Unisim Transformation Summary:
    A total of 95 instances were transformed.
    LUT6_2 => LUT6_2 (LUT5, LUT6): 79 instances
    ...

Here is the schematic view:

[image]

Therefore, if you are creating logic directly, there is not generally a need to create a LUT6_2, and instead just instantiate a LUT5 and a LUT6 and place them onto the BELs directly (e.g. A5LUT and A6LUT).  This is programmatically captured in [Unisim.java](https://github.com/Xilinx/RapidWright/blob/master/com/xilinx/rapidwright/design/Unisim.java#L1220) using the translated prims map.  For example:

`Unisim.LUT6_2.getTransform(Series.Series7);` would be `[LUT6, LUT5]`.

RapidWright currently doesn't have a method to directly instantiate these tranformed prims as the internal connectivity is not fully captured in the framework (yet).  When Vivado writes out the netlist (EDIF), the internals of the LUT6_2 are not written out.  So, in RapidWright, you will have two cells that do not have a corresponding `EDIFCellInst` but will be placed.  

Thoughts?

clavin-xlnx, 2019-02-27:
For now, I will update the error message for transformed prims, to be more accurate:

`ERROR: The unisim LUT6_2 is a transformed prim and cannot be directly instantiated.  Its tranform includes the prim(s) [LUT6, LUT5] which should be used instead.  Look for (or create) an example of the transformed prim in a similar Vivado design for proper connectivity.`

closed 2019-03-01 by clavin-xlnx