Xilinx/RapidWright #25: How to convert a Cell from one Unisim model to another?

mithro, 2019-02-27:
I would like to convert a cell from an async flipflop (`Unisim.FDCE`)  to a sync flipflop (`Unisim.FDCE`).

I was thinking I could create a new cell, copy across all the properties + pin settings and then delete the old cell. However, cells don't seem to have a `listProperties` or similar function? (It has a `getProperty` and `addProperty` functions...)

I can imagine wanting to replace a `Unisim.AND` with `Unisim.LUT2` or `Unisim.LUT4` so it would be nice if there was a helper function for doing this.

clavin-xlnx, 2019-02-27:
the `addProperty()` and `getProperty()` methods on `Cell` are convenience/helper methods that call down to their underlying `EDIFCellInst` equivalents (which are open source).   These convenience methods make code simpler.

I will add a few more helper methods to `Cell` to get properties and set properties but they are maps that use `EDIFName` and `EDIFPropertyValue` objects.

eddieh-xlnx, 2022-02-04:
Closed due to inactivity.

closed 2022-02-04 by eddieh-xlnx