Xilinx/RapidWright #27: Name collsion inside EDIFCell

Db2515, 2019-03-12:
I am trying to run rapid_compile_ipi on the simple IPI design described in the Building Basic Elements for IPI tutorial given on the Xilinx website (https://www.xilinx.com/support/documentation/university/Vivado-Teaching/Digital-Design/2014x/docs-pdf/xup_building_basic_elements_lab.pdf) for the Basys3 board

I am trying to follow the rapid compile work flow so I have not provided a igf file

Synthesis and reading the cache appears to be working however at the block stitcher stages execution stops with the following error message:

N/A | inf | /home/dan/blockCache/e09d51da36316e49/system_xup_and2_0_0_0_routed.dcp
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.esotericsoftware.kryo.util.UnsafeUtil (file:/home/dan/RapidWright/jars/kryo-4.0.0.jar) to constructor java.nio.DirectByteBuffer(long,int,java.lang.Object)
WARNING: Please consider reporting this to the maintainers of com.esotericsoftware.kryo.util.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "main" java.lang.RuntimeException: ERROR: Name collsion inside EDIFCell system, trying to add instance xup_and_vector_0 which already exists inside this cell.
	at com.xilinx.rapidwright.edif.EDIFCell.addCellInst(EDIFCell.java:87)
	at com.xilinx.rapidwright.edif.EDIFCellInst.<init>(EDIFCellInst.java:62)
	at com.xilinx.rapidwright.edif.EDIFCell.createChildCellInst(EDIFCell.java:66)
	at com.xilinx.rapidwright.design.Design.createModuleInst(Unknown Source)
	at com.xilinx.rapidwright.ipi.BlockStitcher.main(BlockStitcher.java:547)

Please let me know if you require any more information, however I am a masters student with exams in the next couple of weeks so please excuse me if am unable to reply till after these.

clavin-xlnx, 2019-03-13:
Thank you for bringing this up.  I was able to reproduce this with the information you provided.  I have a fix that will be available with the next release and I will push a workaround in the meantime once I get it fully tested.  Probably another day or so.

clavin-xlnx, 2019-03-13:
I have pushed a workaround that should resolve this issue--just `git pull` and recompile.  A few other minor bugs popped up while solving this one as it is an extremely small design and the blocks do not interact.  

The workaround will get resolved with our next release, 2018.3.2.  

The issue stemmed from how the EDIFNetlist objects were being merged when creating ModuleInstances.  A recent commit caused a change in behavior that was incompatible with the BlockStitcher flow.  

For reference, I needed to download some of the example files from this page:
https://www.xilinx.com/support/university/vivado/vivado-teaching-material/digital-design.html

clavin-xlnx, 2019-03-28:
2018.3.2 has been released.

closed 2019-03-28 by clavin-xlnx