Xilinx/RapidWright #28: Source code organization for Gradle build

jgoeders, 2019-04-10:
Currently the source code is located in /com, and the gradle.build file indicates this through the statement:

```
sourceSets {
  main {
    java {
      srcDirs = ['com']
    }
}
```

However, this structure isn't quite correct, as ``com`` is part of the source packaging, and ``srcDirs`` should point to the directory containing the source code (ie. one level above ``com``).

This leads to build errors for all of the java files with ``package`` statements:
``The declared package "com.xilinx.rapidwright.debug" does not match the expected package "xilinx.rapidwright.debug"``

Not all build tools are this picky.  For example IntelliJ seems to tolerate this structure (although it seems to be somewhat ignoring the gradle.build file).  If you create a Gradle project in Eclipse, or VSCode, rapidwright will not build correctly.

My first thought was to fix this by changing ``srcDirs`` to ``'.'``, although this doesn't seem to work (I guess it needs to be a subdirectory??)

What does fix this problems is creating a ``src`` folder, and moving ``com`` to this folder, then setting ``srcDirs = ['src']``.   I can do this as a pull request, but I worry about other impacts of this restructuring, so I wanted to get your input Chris.

clavin-xlnx, 2019-04-10:
Thank you for the feedback.  Issue #12 brought up a very similar insight into the code structure, so you are not alone.  Clearly we need to revisit the Gradle file and find the least disruptive solution to both problems.  A change would probably take place with a major release.

jgoeders, 2019-04-11:
Oops, should have checked the open issues closer.  Looks like the exact same issue.

clavin-xlnx, 2019-07-03:
This has been fixed in 2019.1

closed 2019-07-03 by clavin-xlnx