Xilinx/RapidWright #29: Site.getIntTile() produces ArrayIndexOutOfBoundsException

cfib, 2019-04-12:
Site.getIntTile() produces ArrayIndexOutOfBoundsException in line 4 with this Python code
[xc7a35t_getIntTile.py](https://github.com/Xilinx/RapidWright/files/3073249/xc7a35t_getIntTile.py.txt)

I executed RapidWright as follows: 
`java -jar /opt/Xilinx/rapidwright/jars/rapidwright-2018.3.2-standalone-lin64.jar xc7a35t_getIntTile.py.txt`
 
This is the generated traceback: [traceback_getIntTile.txt](https://github.com/Xilinx/RapidWright/files/3073254/traceback_getIntTile.txt)

clavin-xlnx, 2019-04-12:
Thank you for the bug.  It looks like this method was not implemented with Series 7 devices in mind.  I will plan on adding support in the next release.  For now, you can try using this work-around code for SLICEs in Series 7 devices:

```
public static Tile getIntTile(Site s) {
	int wire = s.getTileWireIndexFromPinName("CLK");
	Node n = new Node(s.getTile(),wire);
	for(Wire w : n.getAllWiresInNode()){
		if(Utils.isSwitchBox(w.getTile().getTileTypeEnum())){
			return w.getTile();
		}
	}
	for(PIP p : n.getTile().getBackwardPIPs(wire)){
		Wire w = p.getStartWire();
		for(Wire w2 : w.getNode().getAllWiresInNode()){
			if(Utils.isSwitchBox(w2.getTile().getTileTypeEnum())){
				return w2.getTile();
			}
		}
	}
	return null;
}
```

daveshah1, 2019-05-31:
FYI, I am also seeing a similar exception for certain types of Sites (such as IOBs, which as far as I can tell should have a valid interconnect tile) on the UltraScale+

Minimum test case example:
```java
package dev.fpga.rapidwright;

import com.xilinx.rapidwright.device.Device;

public class bug2 {
    public static void main(String[] args) {
        Device d = Device.getDevice("xczu2cg-sbva484-1-e");
        d.getSite("IOB_X0Y130").getIntTile();
    }
}
```

clavin-xlnx, 2019-07-03:
This should be fixed in 2019.1.0, please reopen if there are further issues.

closed 2019-07-03 by clavin-xlnx