Xilinx/RapidWright #30: rapid_compile_ipi Could not find checkpoint file

Db2515, 2019-04-20:
I am trying to use rapid_compile_ipi on my block design however I am getting the following error message:

ERROR: Could not find file: /home/dan/blockCache/5e87d6aca5f340f9/board_2x2_simple_output_0_1_0_routed_pblock.txt

I have attached the full run log: 
[checker_board_run.txt](https://github.com/Xilinx/RapidWright/files/3100488/checker_board_run.txt)

My project can also be found here: https://github.com/Db2515/FPGA_Rapid_Render

Please let me know if you need other information.

The current design has timing constraint failures which I am working to fix could these be causing the issue? I wanted to post the issue now in case it was caused by something else so I can work on the timing errors in the mean time.

Cheers

clavin-xlnx, 2019-04-23:
I haven't been able to find the board file for your design so I haven't been able to reproduce the error on my side.  However, from the logs, it looks consistent with another problem we were seeing that I fixed just recently.  If you are running Java 9 or later, it produces some warnings that can cause the flow to fail and not produce the DCPs that are needed.  I recently checked in some changes that should fix the problem, not sure if you had them when running this flow.  

You can get that changes, or downgrade to Java 8 (or both) and reset the cache/design using the Tcl command `ultra_clear_cache`.

Questions:
- Which version of Vivado are you using?
- Are you able to provide the board file?  ( I wasn't able to find the exact version you had, maybe it doesn't matter...)
- Which version of RapidWright are you using?

Db2515, 2019-05-03:
Apologies for not replying to this sooner. Making the changes you suggested seemed to work, I think in particular it was pulling the latest version of RapidWright since although I compiled to Java 8 the version of Java within Vivado seems to be Java 9 and I cannot see how to change this. 

Just to answer the questions I am using Vivado 2018.3, I think the board file can be generated using the tcl script found in the bd folder and I am using the 2018.3.3 release of RapidWright

closed 2019-05-03 by Db2515