Xilinx/RapidWright #32: Getting the "tied value" of a node

daveshah1, 2019-05-29:
If you click on a VCC or GND node in Vivado, a "Tied value" item appears in the property pane thus:
[image]

Would it be possible to add a function in RapidWright to get this (if there isn't one already that I've missed?) For now it seems like looking for "VCC_WIRE" or "GND_WIRE" in the node name should work, but this doesn't feel like the cleanest solution.

clavin-xlnx, 2019-05-29:
Yes, this is a good idea.  These attributes also show up using the `report_property [get_nodes <node>]` Tcl command with attributes `IS_GND` and `IS_VCC`.  I'm working on the 2019.1.0 RapidWright release now and I will update the device database files to include this information.

daveshah1, 2019-07-03:
I see that `Node.isTiedToGnd()` and `Node.isTiedToVcc()` functions have been added in 2019.1 and are working for me. Thanks!

closed 2019-07-03 by daveshah1