Xilinx/RapidWright #33: Problems with createIBUFDS

daveshah1, 2019-06-12:
I am trying to add the 125MHz LVDS clock input of a ZCU104 to my design, however I can't get createIBUFDS to work (nor can I find any examples of how to use it).

The following code produces a `NullPointerException`:

```
        Design des = new Design("top", "xczu7ev-ffvc1156-2-e");

        Net clk = des.createNet("clk_i");

        des.createIBUFDS("clk_p", "clk_n",
                des.getDevice().getSite("IOB_X2Y288"), des.getDevice().getSite("IOB_X2Y289"),
                clk, clk.getLogicalNet(), "LVDS");

        des.writeCheckpoint("out.dcp");
```

By some trial and error, replacing the IO standard `"LVDS"` with `null` solves the NullPointerException and produces a dcp, but Vivado finds several problems when opening the DCP:
[image]

As an aside, it would be great to have an example showing how to manually create IO by placing `PAD`, `IBUFCTRL`, etc bels. This would allow exploration of things like MIPI IO and other more specialist features.

clavin-xlnx, 2019-06-12:
Indeed, there is at least one bug in this API and there are no examples of how to use it.  Thank you for bringing this up.  I am working on the 2019.1 release and will get this fixed.  

You make a good point about more granular IO construction.  I am unfamiliar with MIPI IO, but it appears we do [support it in UltraScale+ devices](https://www.xilinx.com/support/documentation/application_notes/xapp1339-mipi-dphy.pdf).  It would seem that MIPI IO configuration is somewhat complex, and might be best to allow Vivado to create the instance and then load it on demand in RapidWright.  If you have a specific application, I could take a look closer.

clavin-xlnx, 2019-06-21:
I recognize I still owe you a fix, but I did want to recognize that the IOB sites you had from your example didn't match what was on the ZCU104 board (at least the latest version of it).  RapidWright does have some APIs to help match package pins to IOB sites, I modified your example code below to illustrate:

```
  Design des = new Design("top","xczu7ev-ffvc1156-2-e");
  Device dev = des.getDevice();
		
  Net clk = des.createNet("clk");
		
  // https://www.xilinx.com/support/documentation/boards_and_kits/zcu104/ug1267-zcu104-eval-bd.pdf, 
  // p.44 Table 3-13
  Site clk125P = dev.getSiteFromPackagePin("H11");
  Site clk125N = dev.getSiteFromPackagePin("G11");
  des.createIBUFDS("clk_p", "clk_n", clk125P, clk125N, clk, clk.getLogicalNet(), "LVDS");

  des.writeCheckpoint("out.dcp");
```

daveshah1, 2019-06-21:
Oops, good catch, looks like I managed to shift the sites down while debugging somehow. Thanks for looking into this!

daveshah1, 2019-06-30:
Having looked into this a bit more, it seems there is also a typo in the manual confusing things (but my initial code wasn't quite correct either). The actual 125MHz clock inputs on my board (ZCU104), also corresponding to the schematics and xdc published by Xilinx, are F23 and E23.

[image]

clavin-xlnx, 2019-07-01:
Thank you for pointing this out.  I have given the feedback to our documentation team and will hopefully get the discrepancy fixed.

closed 2020-03-10 by clavin-xlnx