Xilinx/RapidWright #34: Design.getNetlistPortMap() incorrectly names buses

DallonTG, 2019-06-12:
After I add several ports to a design, I then call Design.getNetlistPortMap(). Some of the ports are bus ports and the map that is returned contains entries with names like "q[15:0][0], q[15:0][1], q[15:0][2]", etc. It seems like the map should instead contain keys for buses like "q[0], q[1], q[2]", etc.

clavin-xlnx, 2019-06-12:
This looks like a bad API that has been leftover from an old project.  It is not really called anywhere and it actually just calls [`EDIFNetlist.generateEDIFPortMap()`](https://github.com/Xilinx/RapidWright/blob/master/com/xilinx/rapidwright/edif/EDIFNetlist.java#L975) to generate the map.  To avoid confusion, I will deprecate this API in the next release, and then remove it after that.  

The code generates a map of all port names for every bit in every port as @DallonTG described but adds an extra bracketed index which is odd and doesn't match common conventions.  

If you'd like to get a map of names -> EDIFPorts, I would suggest calling `EDIFCell.getPortMap()`, in this case on the top cell:

```
		Design d = new Design("example", Device.AWS_F1);
		EDIFCell top = d.getTopEDIFCell(); 
		top.createPort("q[15:0]", EDIFDirection.OUTPUT, 16);		
		for(Entry<String, EDIFPort> e :  top.getPortMap().entrySet()){
			System.out.println(e.getKey() + " -> " + e.getValue());
		}
```
Which outputs:
```
		q -> q[15:0]
```
This map exists so callers don't have to provide bus widths when trying to access bused ports.  Note that if you are trying to connect ports to `EDIFNet` objects, you'll need to instantiate `EDIFPortInst` objects (`EDIFNet` objects connect `EDIFPortInst` objects together and they are only bit-wise).  See the many overloaded method of `EDIFNet.createPortInst(...)`.

DallonTG, 2019-06-13:
Great! Thanks for the explanation & deprecation.

closed 2019-07-03 by clavin-xlnx