Xilinx/RapidWright #35: Nets have no SitePinInsts when placed-only design imported

DallonTG, 2019-06-14:
It is clear that after placement in Vivado, the site pins that need to be routed to/from for nets are known after placement and intra-site routing. However, when I import a placed-only design into RapidWright the site pin collections are not populated for nets. When I import a design that has also been routed however, these site pin collections are populated. This is using Vivado 2019.1 DCPs. Can this be changed so the information is also present for placed-only DCPs?

I also tried a much older version of Vivado - 2016.2. With this version, the site pins are not populated when importing either a placed _or_ routed DCP.

clavin-xlnx, 2019-06-14:
You are correct, RapidWright does not populate `Net` objects with `SitePinInst` objects by default on a placed design.  This is a feature request and has been on my TODO list for a while.  It actually turns out to be a bit tricky and part of the functionality is in the `SiteInst.routeSite()` API.  Since it is not 100% reliable in creating the all the `SitePinInst` objects, I don't call the method when loading the DCP.  

When the DCP already has routing present, it is much easier to identify the missing `SitePinInst` objects (they are not explicitly stored in the DCP).  Another problem with creating them beforehand is that you have a funny situation like this in a SLICE:

[image]

The green LUT can potentially drive two different site pins (see yellow and magenta site wires).  Which one should I instantiate before routing?  You won't really know until the design gets routed.  Of course you could always default to one and switch (or use both), but these are corner cases that need care to be executed properly.

This functionality could be created using the existing APIs in RapidWright and I will have to provide it at some point.  But, I am also open to volunteers to help speed the process along :).

DallonTG, 2019-06-17:
Thanks for the explanation. Like I said, this would be nice functionality to have. If I get around to doing it myself, I'll definitely make a pull request!

clavin-xlnx, 2020-07-31:
I have added a method (`DesignTools.createMissingSitePinInsts(Design design)`) that should satisfy this request.

DallonTG, 2020-08-28:
Looks good! Thanks for doing this.

closed 2020-08-28 by DallonTG