Xilinx/RapidWright #36: Routethru cells not included in Design.getCells()

DallonTG, 2019-06-18:
I would like to be able to get the routethru cells of a design. If I import a DCP with routethru LUTs, they are not included in the list returned by Design.getCells(). However, if I know the site that contains the routethru cells, I can call SiteInst.getCells() and find the routethru cells in that list.

It would be nice to either include the routethru cells in Design.getCells() or to at least update the documentation to specify that routethru cells won't be included.

clavin-xlnx, 2019-06-18:
You are correct, the `Design.getCells()` API does not include routethru cells because Vivado does not provide routethrus in the equivalent Tcl API.  The routethrus are visible in the GUI, so we have `Cell` objects that act as a kind of 'cell placeholder' but aren't treated the same as other Cells.  I can update the Javadoc on that API to make it clear they will not be included.  They are included in a SiteInst.getCells() call, however.

My assumption is that it is not common for people to need to get routethru cells and so there is not a dedicated map lookup for them.  A map could be created by the following code (keyed by SiteName/BELName):

```
Design d = Design.readCheckpoint("your_design_here.dcp");
Map<String,Cell> routethruMap = new HashMap<String, Cell>();
for(SiteInst i : d.getSiteInsts()) {
  for(Cell c : i.getCells()) {
    if(c.isRoutethru()) {
      routethruMap.put(c.getSiteName() + "/" + c.getBELName(), c);
    }
  }
}
```
Feel free to share additional ideas on how to handle routethrus.

clavin-xlnx, 2019-07-03:
2019.1 Javadoc revised, please reopen if further changes are needed.

closed 2019-07-03 by clavin-xlnx