Xilinx/RapidWright #37: getPhysicalPinMapping where the logical pin maps to more than one physical pin

daveshah1, 2019-06-19:
When using a RAMB36E2, control inputs to the cell map to two physical bel pins (for the upper and lower halves) - see the below screenshot.
[image]

`Cell.getPhysicalPinMapping` only returns a single String, and therefore only returns one of the physical pins that the cell pins map to. It is possible to work around this by iterating over `getPinMappingsP2L` and looking for all the physical pins with a given logical pin mapping, but it would be nice to have a function that returned all the physical pins a logical pin maps to.

clavin-xlnx, 2019-06-20:
This is a special case situation and is less common.  I can add an API such as `List<String> Cell.getAllPhysicalPinMappings(String logicalPinName)` if that helps.

daveshah1, 2019-06-20:
Yes, I would find that a useful addition.

daveshah1, 2019-07-03:
Thanks, I can confirm `Cell.getAllPhysicalPinMappings(String logicalPinName)` in RapidWright 2019.1 is working for me.

closed 2019-07-03 by daveshah1