Xilinx/RapidWright #38: SitePinInst.getConnectedCells() produces NullPointerException

jgoeders, 2019-06-20:
The following code produces a NullPointerException on getConnectedCells().  Design dcp/edf attached:

[top_routed.zip](https://github.com/Xilinx/RapidWright/files/3311691/top_routed.zip)

```
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.Net;
import com.xilinx.rapidwright.design.SitePinInst;

public class bug {
    public static void main(String[] args) {
        Design design = Design
        .readCheckpoint("top_routed.dcp");

        for (Net net : design.getNets()) {
            if (net.getSource() == null)
                continue;

            for (SitePinInst sitePinInst : net.getSinkPins()) {
                assert(sitePinInst != null);
                sitePinInst.getConnectedCells();
            }
        }
    }
}
```

jgoeders, 2019-06-20:
I'm using 2018.3

clavin-xlnx, 2019-06-20:
Thanks for reporting this, I believe this API should probably be deprecated and replaced by `DesignTools.getConnectedCells(SitePinInst)` which I have just [committed](https://github.com/Xilinx/RapidWright/commit/a54da72f6e7206745d979287691d1b7bebea8577).  

The problem was that when searching through the site, the code erroneously assumed that the SitePIP would always be non-null.  Let me know if you still see issues after using this new API.

jgoeders, 2019-06-20:
OK, thanks.  

I would like to try that function, but I'm currently on the v2018.3.3-beta tag.  When I switch to the master, the build fails, which I'm assuming is because my jar/data files are from v2018.3.3 and you've been changing APIs in newer commits.  

Is there some way around this?  I'm guessing there aren't "master" versions of the jar/data files...

clavin-xlnx, 2019-06-20:
The latest set of jars/data should generally always work with the latest on master.  I checked the [TravisCI log](https://travis-ci.org/Xilinx/RapidWright/builds/548390394) and it shows a successful compile and runs with the set of jars available for 2018.3.3-beta.  I also downloaded the latest with the jars to double check and it all seems to be working normally, you should just be able to pull, compile and run without problems.

If you are using `gradle` to compile, be sure to set the RAPIDWRIGHT_PATH variable to your project directory.  If you still have issues, I would recommend against using gradle, at least until the next release.  I would try Eclipse/IntelliJ or just plain javac to compile (for example: `javac find . -name '*.java'` from the RapidWright directory).

jgoeders, 2019-06-21:
OK, I'm not sure what the problem was.  I'll give it another shot.  Thanks for linking to the TravisCI log.

clavin-xlnx, 2019-07-03:
Closing this, if build problems persist from updating, please open a new issue.

closed 2019-07-03 by clavin-xlnx