Xilinx/RapidWright #39: SitePinInst.getConnectedCells() produces unexpected results

jgoeders, 2019-06-20:
(v2018.3)
I'm trying to use SitePinInst.getConnectedCells() on the following design:

[top_routed.zip](https://github.com/Xilinx/RapidWright/files/3311811/top_routed.zip)

The code below prints the following:

> aes128_0/r2/t0/t3/s4/g3_b2
> aes128_0/r2/t0/t3/s4/out[2]_i_1
> aes128_0/r2/t0/t3/s4/g0_b2
> aes128_0/r2/t0/t3/s4/g1_b2
> aes128_0/r2/t0/t3/s4/g3_b2

One issue is that there appear to be duplicates listed (not a big issue).  The other issue is that I don't see how the D6 pin is connected to the listed cells.  I've included a Vivado screenshot of the design, with the Pin marked.  The cells are highlighted in Purple.  

[image]

```
import com.xilinx.rapidwright.design.Cell;
import com.xilinx.rapidwright.design.Design;
import com.xilinx.rapidwright.design.SiteInst;
import com.xilinx.rapidwright.design.SitePinInst;

class bug2 {
    public static void main(String [] args) {
        Design design = Design
                .readCheckpoint("top_routed.dcp");

        SiteInst siteInst = design.getSiteInst("SLICE_X50Y85");
        assert(siteInst != null);
        SitePinInst sitePinInst = siteInst.getSitePinInst("D6");
        for (Cell cell : sitePinInst.getConnectedCells()) {
            System.out.println(cell.getName());
        }        
    }
}
```

clavin-xlnx, 2019-06-20:
I've just [committed](https://github.com/Xilinx/RapidWright/commit/a54da72f6e7206745d979287691d1b7bebea8577) an alternative to this API, `Design.getConnectedCells(SitePinInst)` that resolves this issue.  It returns a set instead of a list to avoid duplicate situations.  I will deprecate and eventually remove the API on `SitePinInst`.  

The problem was caused by the fact that the site wire `D6` not only drives `A6` on the `D6LUT` but it also connects to pins `WA6` on all 4 of the lower LUTs (A6LUT, B6LUT, ...).  The solution is to also check the connecting cell if it has a pin mapping for that particular physical pin, if it does not, then it is likely not connected.  

Again, let me know if you see other issues.

jgoeders, 2019-06-20:
Thanks! Yes I did just figure out that D6 is connected to the other pins, and it was a bit of misunderstanding on my point, as I am looking for what is connected in the design (through routing), not what is connected in the device.  

> The solution is to also check the connecting cell if it has a pin mapping for that particular physical pin, if it does not, then it is likely not connected.

OK, I'll try this.  This should tell me if the intra-site routing connects the site pin to the cell?

jgoeders, 2019-06-20:
Oh never mind my last message.  I see that your new function does this exact thing and returns the correct set of cells.  Thanks!!

closed 2019-07-03 by clavin-xlnx