Xilinx/RapidWright #42: EDIFCellInst.equals() returns True for instances of different type in different parents

cfib, 2019-07-25:
This might not be a bug, just a documentation issue but certainly was a pitfall for me:

The minimal netlist `netlist.edf` in the attached [CellInsts.zip](https://github.com/Xilinx/RapidWright/files/3430767/CellInsts.zip) consists of a toplevel module that instantiates two modules `lut_a_container` and `lut_b_container`. These, in turn, instantiate a `LUT1` and a `LUT2`, respectively. Each LUT is instantiated under the name `lut_inst`.

After running

    lut_in_container_a = netlist.getCellInstFromHierName('lut_a_container_inst/lut_inst')
    lut_in_container_b = netlist.getCellInstFromHierName('lut_b_container_inst/lut_inst')

I receive the following results (see `rapidwright.py` in the attached ZIP):

    lut_in_container_a.equals(lut_in_container_b)
	    True
    lut_in_container_a == lut_in_container_b
	    True
    lut_in_container_a is lut_in_container_b
	    False

To avoid this potential pitfall (these two instances may have the same EDIFName, but are different instances), IMO either the `EDIFCellInst.equals()` operator should take the parent cell into account or the documentation should reflect this behavior.

clavin-xlnx, 2019-07-25:
This looks to be a bug, thank you for pointing this out.  I agree, the cell type should most certainly be included in checking for equality among two EDIFCellInst.  I just surveyed the EDIF objects and how they implement equality as this may have not been fully examined previously:

| EDIF Object | Equality check | 
|--------------|-----------------|
| `EDIFCell`|`EDIFName.equals()`|
| `EDIFCell`|`EDIFName.equals()`|
| `EDIFCell`|`EDIFName.equals()`|
| `EDIFCell`|`EDIFName.equals()`|
| `EDIFCell`|`EDIFName.equals()`|

clavin-xlnx, 2019-07-25:
My apologies, I just noticed you suggested the parent type of the `EDIFCellInst` should be checked.  That information is more contextual of the `EDIFCellInst` and does not necessarily have bearing on what the `EDIFCellInst` is.  But I do agree that the documentation should be made more clear on the subject in any case.