Xilinx/RapidWright #45: How to use blockplacer to place a design without specifying the location

hpan5, 2019-08-31:
I'm trying to use plockplace to create a LUT without specifying the location

`    
    EDIFCell or2Cell = netlist.getHDIPrimitive(Unisim.OR2);
    EDIFCellInst or2Inst = top.createCellInst("or2Inst", or2Cell);
    Cell or2 = design.createCell("or2", or2Inst); 
`

and use blockplacer to place and route the design
`
    Design d = new BlockPlacer2().placeDesign(design, false);
    d.routeSites();
    new Router(d).routeDesign();
`
However, I got the following synthesis errors when I was trying to run the program:

`WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.esotericsoftware.kryo.util.UnsafeUtil (file:/mnt/c/Users/27688/Documents/Git/rapidwright-2019.1.1-standalone-lin64.jar) to constructor java.nio.DirectByteBuffer(long,int,java.lang.Object)
WARNING: Please consider reporting this to the maintainers of com.esotericsoftware.kryo.util.UnsafeUtil
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "main" java.lang.NullPointerException
	at com.xilinx.rapidwright.design.Net.connect(Unknown Source)
	at autoplaced_or_lut.main(autoplaced_or_lut.java:64)
`
What's the right way to use blockplace to place the design? Any hints and suggestions?

My edition of Lesson1.java can be found here: 
[autoplaced_or_lut.zip](https://github.com/Xilinx/RapidWright/files/3562778/autoplaced_or_lut.zip)

clavin-xlnx, 2019-09-05:
The Block Placer (`BlockPlacer2`) is specifically built to place `ModuleInst` ojbects (instances of `Module` objects).  These are generally smaller designs that have been placed and routed out of context of a full design.  The fine grained placement and routing problem has already been decided and thus are reused by the Block Placer when moving all the cells as a contiguous unit.  

It sounds like what you want is a general placement tool which RapidWright currently doesn't have yet.  In order to place the LUT by `BlockPlacer2` automatically, you would need to instantiated it in a separate design, place it, encapsulate it inside a `Module` and then create an instance of it (a `ModuleInstance`) in the design.  This may not be the answer you are looking for, but I will attach a code example soon that illustrates how this could be done, although it may not be the best way to proceed.

clavin-xlnx, 2019-09-06:
I have made a few minor changes to RapidWright and pushed them today.  If you pull the latest changes and run the attached code ([autoplaced_or_lut.zip](https://github.com/Xilinx/RapidWright/files/3585156/autoplaced_or_lut.zip)), you should be able to achieve what you were expecting.  It is a bit more tedious because the block placer only operates on module instances and so you have to package up the LUT (OR2) into a module and then create the `ModuleInst` of it in the design.  

Typically, a `Module` and `ModuleInst` are defined by designs that come directly from Vivado, but you can generate them from scratch--you just have to be consistent in both the logical and physical creation of the netlist.

closed 2019-10-04 by clavin-xlnx