Xilinx/RapidWright #49: About ModuleInst Connection: What if some ports in the ModuleInst have no corresponding Pins?

LJChen98, 2019-09-16:
Hi, 
Now, I have a pre-implemented module, some ports of which have no corresponding pins, just like port `dataInArray_0` below:
[image]

When I was trying to use the method `ModuleInst.connect()` to make the connection between port `dataInArray_0` and other port of other moduleInst, some errors would occur. 

I looked into the code of `ModuleInst.connect()`, it seems that the code doesn’t take this situation into consideration, which makes “Connect physical pins” a compulsory process. However, some ports have no corresponding pins, let alone making connections.

I am not sure whether or not it is a bug. However, referring to what the code do to “Connect to a top-level port”, I suppose this problem can be solved by ignoring “Connect physical pins” when a port has no corresponding pin.

I would appreciate it if anyone could help me examine this problem.

clavin-xlnx, 2019-09-16:
I believe the `ModuleInst.connect()` method may need further effort to make it more robust in these kinds of situations.  It should make connections even if there is no physical logic behind the port.  

Would you be able to provide an example test case design and/or code to illustrate a test case of the issue?

LJChen98, 2019-09-17:
Hi, 
An example test case is attached. [test.zip](https://github.com/Xilinx/RapidWright/files/3618944/test.zip)

I suppose ignoring “Connect physical pins” when a port has no corresponding pin could solve this problem, which is also actually what the code do, if true, when connecting a topper-level port to a ModuleInst port. 

I fix this myself by adding `if (pin0 == null || pin1 == null) return;` after getting  corresponding pin. Until now, It seems to work well with no problem, but I am not sure if this modification would cause some other problems because I am not familiar with what the RapidWright underlying code do to a dcp file...

Thanks.

eddieh-xlnx, 2022-02-04:
Closed due to inactivity.

closed 2022-02-04 by eddieh-xlnx