Xilinx/RapidWright #50: updateProbesConnections() causes a NullPointerException

TailoredWisdom, 2019-09-16:
Hi,

I have an issue with one of the tutorials that may be related to the code.

When following the ProbeRe-router tutorial in the jupyter notebook, invoking `ProbeRouter.updateProbeConnections(design,probesMap)` a NullPointerException.
It has been tested on Windows and Linux, in the jupyter notebook and on Rapidwright itself.
Here is the output:

    Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
	at com.xilinx.rapidwright.edif.EDIFNetlist.getNetAliases(EDIFNetlist.java:740)

	at com.xilinx.rapidwright.edif.EDIFNetlist.generateParentNetMap(EDIFNetlist.java:881)

	at com.xilinx.rapidwright.edif.EDIFNetlist.getParentNetMap(EDIFNetlist.java:827)

	at com.xilinx.rapidwright.edif.EDIFNetlist.getPhysicalNetFromPin(EDIFNetlist.java:520)

	at com.xilinx.rapidwright.debug.ProbeRouter.updateProbeConnections(ProbeRouter.java:90)

	at com.xilinx.rapidwright.debug.ProbeRouter.updateProbeConnections(ProbeRouter.java:70)

	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)

	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)

	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)

	at java.base/java.lang.reflect.Method.invoke(Method.java:567)

    java.lang.NullPointerException: java.lang.NullPointerException

I have taken a look at the code, unfortunately I am not familiar with Java. I suspect it may related to `opr` or `otherNet`, but I wouldn't be able to tell.

I would appreciate if anyone could take a look into this.

clavin-xlnx, 2019-09-16:
Thank you for bringing this up, I believe this is due to a recent change ([359951d](https://github.com/Xilinx/RapidWright/commit/359951dc75850ac18a53272d7d8da06f5543b610)) that was fixing another issue and didn't fully capture an NPE potential.  I have committed a fix just now and this should be resolved.

TailoredWisdom, 2019-09-17:
Thank you, this indeed fixes the issue.

closed 2019-09-17 by TailoredWisdom