Xilinx/RapidWright #56: EDIF Parser fails on submodules with certain characters in their names

jakobwenzel, 2020-02-07:
When reading a DCP that contains a submodule with certain characters (including \ and $) in its name, the EDIF Parser fails:

```
java.lang.Exception
	at com.xilinx.rapidwright.edif.EDIFParser.expect(EDIFParser.java:733)
	at com.xilinx.rapidwright.edif.EDIFParser.parseEDIFCell(EDIFParser.java:545)
	at com.xilinx.rapidwright.edif.EDIFParser.parseEDIFLibrary(EDIFParser.java:471)
	at com.xilinx.rapidwright.edif.EDIFParser.parseEDIFNetlist(EDIFParser.java:389)
	at b.a(Unknown Source)
	at com.xilinx.rapidwright.design.Design.readCheckpoint(Unknown Source)
Parsing Error: Expected token: (, encountered: rename on line: 79.
```

The DCP was created from this Verilog code:

```verilog
`default_nettype none
module top (input wire i , output wire o, input wire clk);
	\evil\$characters inst (i,o,clk);
endmodule

module \evil\$characters (input wire i, output reg o, input wire clk);
	always @(posedge clk) o <= i;
endmodule
```
Using this TCL code on Vivado's console:
```tcl
read_verilog "design.v"
synth_design -flatten_hierarchy none -top top -part XC7A200T-SBG484-1
write_checkpoint -force design.dcp
```

clavin-xlnx, 2020-02-07:
Thanks for reporting this!  It appears to be a problem with the EDIF field in `cell` called `view`.  This test case shows that the `view` qualifies for the same treatment as all other EDIF naming constructs and I can hopefully just update the code with the appropriate handler to address the renaming.  

Surprising that we haven't come across this case yet, so glad you pointed it out.  I'm working on a fix, hope to have it tested/committed soon.

jakobwenzel, 2020-02-21:
Thanks for the fix! Works perfectly for my larger designs

closed 2020-02-21 by jakobwenzel