Xilinx/RapidWright #64: Using deprecated code QApplication.setGraphicsSystem("raster");

ilesser, 2020-06-01:
The method `setGraphicsSystem` was removed in newer versions. 

See https://doc.qt.io/qt-5/sourcebreaks.html

```
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/device/browser/DeviceBrowser.java:88: error: cannot find symbol
		QApplication.setGraphicsSystem("raster");
		            ^
  symbol:   method setGraphicsSystem(String)
  location: class QApplication
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/device/browser/DeviceBrowser.java:88: error: cannot find symbol
		QApplication.setGraphicsSystem("raster");
		            ^
  symbol:   method setGraphicsSystem(String)
  location: class QApplication
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/device/browser/DeviceBrowserScene.java:131: warning: [deprecation] Integer(int) in Integer has been deprecated
				i = new Integer(1);
				    ^
/home/ilesser/fpga/Xilinx/RapidWright/jars/qtjambi-4.5.2_01-javadoc.jar(/com/trolltech/qt/QNativePointer.java):884: warning: [deprecation] finalize() in Object has been deprecated
    protected void finalize() {
                   ^
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/placer/handplacer/HandPlacer.java:96: error: cannot find symbol
		QApplication.setGraphicsSystem("raster");
		            ^
  symbol:   method setGraphicsSystem(String)
  location: class QApplication
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/placer/handplacer/HandPlacer.java:117: error: cannot find symbol
		QApplication.setGraphicsSystem("raster");
		            ^
  symbol:   method setGraphicsSystem(String)
  location: class QApplication
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/placer/handplacer/HandPlacer.java:96: error: cannot find symbol
		QApplication.setGraphicsSystem("raster");
		            ^
  symbol:   method setGraphicsSystem(String)
  location: class QApplication
/home/ilesser/fpga/Xilinx/RapidWright/src/com/xilinx/rapidwright/placer/handplacer/HandPlacer.java:117: error: cannot find symbol
		QApplication.setGraphicsSystem("raster");
		            ^
  symbol:   method setGraphicsSystem(String)
  location: class QApplication

```

clavin-xlnx, 2020-06-02:
Unfortunately, RapidWright is using an old Java port of Qt called Qt Jambi which appears to no longer be under active development.  There are a few newer builds--which have some compatibility issues, but none approached Qt 5.  Ultimately, it would be best to switch to an actively developed graphical library ([JavaFX](https://openjfx.io/) perhaps?) but that would require some non-trivial re-work of some RapidWright GUIs.

ilesser, 2020-06-02:
My knowledge of Java and gui apps is tangencial.

To be honest I simply removed the piece of deprecated code and compiled rapidwright to use it under nextpnr-xilinx. Removing the set Graphics System did the trick.

I compiled it using the default library libqt5opengl5 under Linux mint 19.1 Tessa (Ubuntu 18.04 lts).
Is there a reason why we need to use qt jambi or Java fx instead of libqt5opengl5?

Sorry that I can't be of much help here.

clavin-xlnx, 2020-06-05:
I see the issue when compiling nextpnr-xilinx.  Java Qt Jambi is currently a dependency to serve as a bridge between Java and C++ Qt.  Unfortunately, there isn't a Qt 5 version of Qt Jambi that we can upgrade to. 

Perhaps there is a way to move to Qt5 but using a different bridge (Bytedeco's Java bridge: https://github.com/bytedeco/javacpp-presets).  This could be added to the TODO pile and see if it could be used port away from Qt Jambi.

eddieh-xlnx, 2022-02-04:
Currently, there are no plans to support Qt5 in the future, and as stated in https://github.com/Xilinx/RapidWright/pull/65#issuecomment-637793785 it looks like removing this call for Qt4 dramatically affects performance on Linux platforms.

closed 2022-02-04 by eddieh-xlnx