Xilinx/RapidWright #67: Incomplete Properties of Cells

btut, 2020-06-23:
Hi,
I noticed that I cannot access most properties of EDIFCellInsts. getProperty just results in an empty Map. I know there are lots of properties, both standard properties and properties I created myself, as I can read them in Vivado.

clavin-xlnx, 2020-06-23:
Your observations are correct, only non-default properties are set/stored on the cell instance.  RapidWright does not (yet) have the set of default settings on cells/cell instances.  You can assume they are implicit, but I agree it would be nice to have them accessible.

I will look into the effort needed to provide this information in a future release.

btut, 2020-06-23:
I don't understand. What do you mean by non-default? I can't see any properties. No matter if the values are the default ones or not. And what's most important to me, I can't see custom properties created by me at all. Is that what you mean by non-default?

clavin-xlnx, 2020-07-31:
(Apologies in the large delay to respond, I thought I had already responded here.)

The EDIFCellInst stores properties that are available in the EDIF file itself.  The absence of any properties means that the cell is configured with default values as can be seen on the cell in Vivado.  If any properties are set away from those default settings, an entry is made in the property map and can also be examined in the EDIF file when exported by Vivado.  

Ideally, the primitive cells have a know set of properties and this information could be added to RapidWright, but it is currently not present. 

You should be able to add custom properties to EDIFCellInsts by adding them directly, for example:

```
Design d = new Design("HelloWorld",Device.PYNQ_Z1);
Cell and2 = d.createAndPlaceCell("and2", Unisim.AND2, "SLICE_X100Y100/A6LUT");
and2.getEDIFCellInst().addProperty("MyCustomProperty", "MyCustomValue");
System.out.println(and2.getEDIFCellInst().getProperties());
```
This should output:
```
{INIT=string(4'h8), MyCustomProperty=string(MyCustomValue)}
```
Here `INIT` is a property that is configured to the circumstance (AND2 behavior) and also the `MyCustomProperty` can also be found.

eddieh-xlnx, 2022-02-04:
Closed due to inactivity.

closed 2022-02-04 by eddieh-xlnx