Xilinx/RapidWright #74: cell creation from existing cell

shivnandanpandey, 2020-09-09:
Hi,
i am trying to replicate a cell from post place dcp with following code 
`name = String.format(cell.getName() + "_dup");
nameInst = String.format(cell.getEDIFCellInst().getName() + "_dup"); 
EDIFCellInst cellI = cell.getParentCell().createChildCellInst(nameInst, cell.getEDIFCellInst().getCellType());
newCell = cell.copyCell(name, cellI);
design.addCell(newCell);`

in most of the cases it is working fine but in some case , it is showing error while reading post rapid wright dcp in vivado for routing.

ERROR: [EDIF 20-84] Cannot find instance 'S00_AXI_RDATA_255__INST_0_i_1_dup' in view 'axi_256_to_64_axi_interconnect_v1_7_16_fifo_gen'. [../post_place_rw/testBench_0.edf:201970]

Could you please help me to replicate a cell in correct manner?

Thanks,
Shivnandan

clavin-xlnx, 2020-09-09:
For the most part, the above approach seems reasonable.  However, to really debug what might be going on, you may need to provide a test case (minimal code and DCP) that exhibits the behavior.

One consideration is that when you modify the logical netlist cell (`EDIFCell`), if there is more than once instance in the design that instantiates that particular cell, you'll be changing the design in multiple places.  If your design came directly from Vivado, this probably isn't an issue because it creates unique cells for each instance.

Another consideration is that there are macro primitives that expand and collapse when reading/writing a DCP/EDIF.  In the EDIF, the leaf cells are hidden, but upon load in RapidWright or Vivado, the macro primitives are expanded into separate leaf cells.  An example of this is a LUT6_2.  When written out to EDIF, it is a black box, however upon load into RapidWright or Vivado, the leaf cells LUT5 and LUT6 are defined.  If you are copying one of the LUT5 or LUT6 instances, you'll probably run into an issue.

shivnandanpandey, 2020-09-16:
Hi Clavin,

I tried to reproduce the issue on small test case but couldn't make it.

I have attached the problematic cell schematic. 
Candidate cell is high fan out cell, i am trying to do cell replication to reduce the fan out.
When i am just creating the cell i am not getting any error. but when i am trying to put the net connection, i am getting mentioned error while loading the dcp in vivado.

For load(sink) net i am using following code to create net and connect the connection. netSink is original load net. 
`EDIFNet net = netSink.getParentCell().createNet(name);`
for driver net ..
`netDriver.createPortInst(port, cellIn);`     cellIn is new dup cell's  edif cell inst.

In most of the cases it is working fine. but in some cases it is giving error.
  
Thanks,
Shivnandan 
[image]

shivnandanpandey, 2020-09-17:
I have identified the issue.
In some designs, for few cells, the parent cell of the physical cell and parent cell of the logical cells are not the same . in this situation, we are facing this issue.
Currently, I am checking this and skipping those cells.

clavin-xlnx, 2022-02-04:
Closing due to inactivity.

closed 2022-02-04 by clavin-xlnx