Xilinx/RapidWright #75: vu19p and vu15p availability

shivnandanpandey, 2020-09-16:
Hi 

I was checking the supported devices list in device data of package.
I was looking for vu15p and vu19p  devices. i was not able to find them.

Could you please let me know when can we expect the addition of these devices?

Thanks,
Shivnandan

clavin-xlnx, 2020-09-16:
When those devices are generally available in the latest release of Vivado, they will be made available in the following release of RapidWright, probably a month or so after the Vivado release.

clavin-xlnx, 2020-11-02:
Quick update: The VU19P is now available in the [2020.1.4 Release](https://github.com/Xilinx/RapidWright/releases/tag/v2020.1.4-beta).  We are still waiting for public access for the VU15P before it can be added to RapidWright.

closed 2022-02-04 by clavin-xlnx