Xilinx/RapidWright #76: In placed design, adding some cells and placing them

shivnandanpandey, 2020-09-17:
Hi,

I have added some cells in placed design, is there any optimal way to place them all in Rapid wright only?
I know, there are API's to place them one by one by giving the loc information.

Basically, I am doing cell replication. Here optimal means, I want to place them in between the loads and the driver of the replicated cell. 

Thanks,
Shivnandan

clavin-xlnx, 2020-09-18:
There are plans to develop a general purpose placer for RapidWright that can cover the case you have described, unfortunately, that is still in the works.  

There are enough resources to build a special-purpose method that could accomplish this task directly.  Take a look at `DesignTools.placeCell()`.  This is just a quick and dirty placement of a cell.  What I believe is you would want to find the centroid (center point of all pins involved in the new net).  There is some code in RapidWright used for the `BlockPlacer` that could be useful for doing so.  See lines [383-384 in the class BlockPlacer](https://github.com/Xilinx/RapidWright/blob/c97b120ab0b3f426b9a79e64af2f5d58b884a321/src/com/xilinx/rapidwright/placer/blockplacer/BlockPlacer.java#L383), there is a call to a method called `SmallestEnclosingCircle.getCenterPoint()` that takes as input a list of coordinates (your source and sink pin tile coordinates).  and it will return the center coordinate (ideal location for a pipeline flop).  

You could then use a [PriorityQueue](https://docs.oracle.com/javase/7/docs/api/java/util/PriorityQueue.html) with a customized comparator (minimized distance to centroid) that is loaded with a window of neighboring sites with the closest falling to the bottom first.  You could then iterate over each of those sites until you find a free BEL to place the flop.

clavin-xlnx, 2022-02-04:
Closing due to inactivity.

closed 2022-02-04 by clavin-xlnx