Xilinx/RapidWright #78: Getting incorrect Driver Cell for the EDIFNet

braju4280, 2020-09-22:
[image]

Hi calvin,
I am trying to get driver cell for EDIFNet, used the following code to get driver cell for EDIF net(net_a_0), It gives the wrong driver cell (i.e., Ub_2/gen[4]), it should have return  “Ua_0/gen[4]”. For most of the nets i am able to get correct results.  Is this correct way to get driver cell for above representation(shown in figure)? I am not sharing my design since it is confidential. Please let me know also how can i share my design confidentially.

`List\<EDIFPortInst\> src = eNet.getSourcePortInsts(false);           // Here eNet is EDIFNet “net_a_0”
System.out.println(src.get(0).getFullName());                             // prints  Ua_0/Ia[4]           
EDIFNet internalNet = src.get(0).getInternalNet();                     // returns Ia[4]        
                        
List\<EDIFPortInst\> srcIn = internalNet.getSourcePortInsts(false);
System.out.println(srcIn.get(0).getFullName());                         //prints gen[4]/Q
System.out.println(srcIn.get(0).getCellInst().getName());          //prints Ub2/gen[4]`

clavin-xlnx, 2020-09-22:
Happy to help, feel free to email me (listed under my account) and we can arrange a private design share.  

The behavior of the final line in your example is strange, although I'm not sure if it should return "Ua_0/gen[4]".  The way RapidWright and EDIF represent the netlist is through a ["folded netlist"](https://en.wikipedia.org/wiki/Netlist#Hierarchy) representation.  That is, if two cell instances instantiate the same cell, they share the library definition.  In this case, the EDIFPortInst is inside the cell definition of Ua_0 and doesn't carry its context information as it can be instantiated multiple times.  I would have expected just "gen[4]" as being the cell instance name for portInst Q rather than Ub_2/gen[4].  I ran your code on a local design which behaves as expected.  

To manage the folded netlist representation, we use `EDIFHierPortInst`, `EDIFHierNet` and `EDIFHierCellInst` classes which keep the context (full hierarchical name) of the current object in view.  Although I don't have an example of your attempt here, a related method that finds the sinks from a net is found in [`EDIFNetlist.getSinksFromNet(EDIFHierNet net)`](https://github.com/Xilinx/RapidWright/blob/c97b120ab0b3f426b9a79e64af2f5d58b884a321/src/com/xilinx/rapidwright/edif/EDIFNetlist.java#L1019)

It seems there is a discrepancy in expected behavior, so a concrete design might help remedy the issue.

clavin-xlnx, 2022-02-04:
Closing for now, please reopen if further questions arise.

closed 2022-02-04 by clavin-xlnx