Xilinx/RapidWright #82: setting  the constraints on a cell.

shivnandanpandey, 2020-10-07:
Hi,

I am trying to replicate a register. code is working fine but I am facing a timing issue.
I couldn't find a way to set (copy) the constraint from the original to the newly created cell. this is resulting in very bad QoR.

could you please help me to resolve this issue?

Thanks,
Shivnandan

clavin-xlnx, 2020-10-07:
Hi Shivnandan,

Could you describe the timing constraint you had on the original register?  A clock constraint, assuming you also route the same clock to the register should propagate automatically when loaded in Vivado.  

If you need to add another constraint because you've added more logic to the design, you can do so by using the XDC APIs in the `Design` class to add additional XDC constraints that would now apply to the new register.  For example, if the additional register adds a new multicycle path, you might add additional XDC constraints as follows:

```
Design design = Design.readCheckpoint(...);
...
design.addXDCConstraint("set_multicycle_path 2 -setup -from [get_pins source_reg/C] -to [get_pins new_replica_reg/D]");
design.addXDCConstraint("set_multicycle_path 1 -hold -from [get_pins source_reg/C] -to [get_pins new_replica_reg/D]");
...
```

There is a chance that Vivado cannot close timing properly because the netlist is malformed.  If this is the case, then no constraint will resolve the issue.  The netlist creation must be corrected first.  I am happy to examine any examples if you are able to share. 

Chris

clavin-xlnx, 2021-11-18:
Closing for now, please re-open if needed.

closed 2021-11-18 by clavin-xlnx