Xilinx/RapidWright #89: getUsedSitePIPs() for OPMODEx/OPMODEx_B in DSP48E1

cfib, 2020-10-23:
Hi! I'm trying to extract the SitePIPs related to the `OPMODE` signal used by `DSP48E1` in a placed & routed design.

Vivado shows that for `OPMODE[4]` and `OPMODE[5]` the `OPMODE4_B` and `OPMODE5_B` signals are used, while `OPMODE[6]` uses `OPMODE6`

[image]

In Rapidwright (Version 2020.1.3 Beta), I use `siteInst.getUsedSitePIPs()` for the DSP48E1 site instance, which reports:

    ALUMODE0INV.ALUMODE0->>OUT
    ALUMODE1INV.ALUMODE1->>OUT
    ALUMODE2INV.ALUMODE2->>OUT
    ALUMODE3INV.ALUMODE3->>OUT
    CARRYININV.CARRYIN->>OUT
    CLKINV.CLK->>OUT
    INMODE0INV.INMODE0->>OUT
    INMODE1INV.INMODE1->>OUT
    INMODE2INV.INMODE2->>OUT
    INMODE3INV.INMODE3->>OUT
    INMODE4INV.INMODE4->>OUT
    OPMODE0INV.OPMODE0->>OUT
    OPMODE1INV.OPMODE1->>OUT
    OPMODE2INV.OPMODE2->>OUT
    OPMODE3INV.OPMODE3->>OUT
    OPMODE4INV.OPMODE4->>OUT
    OPMODE5INV.OPMODE5->>OUT
    OPMODE6INV.OPMODE6->>OUT

For `OPMODE4` and `OPMODE5` I expected to see:

    OPMODE4INV.OPMODE4_B->>OUT
    OPMODE5INV.OPMODE5_B->>OUT

Am I misunderstanding `getUsedSitePIPs()`? 

The used netlist, DCP, EDF, and scripts are attached: [mwe.zip](https://github.com/Xilinx/RapidWright/files/5428569/mwe.zip)

clavin-xlnx, 2020-10-28:
You are correct in your understanding, unfortunately, the behavior of Vivado is not always as consistent given this paradigm.  It appears that on some inverting BELs that the SitePIP is secondary to the nets mapped to the inputs/outputs of the BEL.  In this case, the input to OPMODE6INV is GND, however OPMODE5INV and OPMODE4INV have VCC as the input.  Since all three have outputs mapped to the GND net, the inverting function is only active in OPMODE4INV and OPMODE5INV and overrides the SitePIP settings:

[image]

I recognize this is not as intuitive and not a satisfying behavior.  Perhaps you could share more about what you use case is and could suggest an alternative approach.

cfib, 2020-11-11:
In essence, I am trying to extract a post-placement netlist of a design that contains all wire segments (and therefore INT PIPs and Routing BELs with their configuration).

clavin-xlnx, 2020-11-11:
Yes, in that case, this behavior will appear inconsistent.  There is another developing project that aims to enable easier import/export of placed and/or routed netlists: https://interchange.vtr.tools.  There is an experimental 'interchange' branch in RapidWright that is working to implement this specification.  Ultimately, not quite ready for prime time, but it will have to handle the issue you have mentioned.

cfib, 2020-11-12:
Thank you, I'll take a look at that,

clavin-xlnx, 2022-02-04:
Closing for now, please reopen if there are further questions.

closed 2022-02-04 by clavin-xlnx