Xilinx/RapidWright #96: How to merge two routed DCPs?

Licheng-Guo, 2020-11-12:
Hi Chris,

I'm trying to merge two DCPs, however, the placement & routing information gets lost. Do you know where I did wrong? Thanks a lot!!!

```java
public static void main(String[] args) {
    Design design = new Design("merge", Device.PYNQ_Z1);
    design.setAutoIOBuffers(false);
    design.setDesignOutOfContext(true);
    Device device = design.getDevice();
    EDIFNetlist netlist = design.getNetlist();
    EDIFCell top = netlist.getTopCell();

    Design p1 = Design.readCheckpoint("typicalRoute.dcp"); // contains two FDREs connected by a Net
    Design p2 = Design.readCheckpoint("typicalRoute2.dcp"); // contains another two FDREs connected by a Net
    
    netlist.migrateCellAndSubCells(p1.getNetlist().getTopCell());
    netlist.migrateCellAndSubCells(p2.getNetlist().getTopCell());

    ModuleInst m1 = design.createModuleInst("p_1", new Module(p1));
    ModuleInst m2 = design.createModuleInst("p_2", new Module(p2));

    design.writeCheckpoint("merge.dcp");
}
```

In the final DCP, I could see p1 and p2 in the netlist, but the leaf cells become unplaced. I have not been able to figure this out from the available examples. Thanks a lot for your help!

Best,
Licheng

clavin-xlnx, 2020-11-12:
Hi Licheng,

Seems like you have setup things right, you just need to declare where you want the ModuleInsts to go.  You can use ModuleInst.place(Site) API to do so, for example:

```
Site m1Placement = device.getSite("SLICE_X10Y10");
if (!m1.place(m1Placement)){
  System.err.println("ERROR: Couldn't place m1 on " + m1Placement + ", please choose a valid location.");
}
Site m2Placement = device.getSite("SLICE_X10Y15");
if (!m2.place(m2Placement)){
  System.err.println("ERROR: Couldn't place m2 on " + m2Placement + ", please choose a valid location.");
}
```
This will relocate the original placement and any routing from the original design/Module.  If the desired location is incompatible with the provided site, the API will return false.

Chris

Licheng-Guo, 2020-11-12:
That works! Thanks a lot!

This tool is really helpful. We greatly appreciate your effort!

Licheng-Guo, 2020-12-04:
Hi Chris,

I encountered another related issue. Continuing the previous example with the following code:
```java
System.out.format("m1 has anchor %s at %s\n", m1.getAnchor().getName(), m1.getAnchor().getSiteName() );
System.out.format("m2 has anchor %s at %s\n", m2.getAnchor().getName(), m2.getAnchor().getSiteName() );
```

What will be printed out is:
```
m1 has anchor p1/SLICE_X29Y80 at null
m2 has anchor p2/SLICE_X30Y71 at null
```

Similarly, m1.getAnchor().getInstanceX() and getInstanceY() will return -1.

I wonder what I did wrong? I just need to place the ModuleInst at the location where the original DCP is placed.

Thanks for your help!

Best,
Licheng

clavin-xlnx, 2020-12-04:
Hi Licheng,

The behavior you are describing seems consistent with the `ModuleInst` being in an unplaced state.  You can double check this by calling `ModuleInst.isPlaced()`.  When you attempt to place the `ModuleInst`, does the `ModuleInst.place()` API return true?  Feel free to share more of the code/example if you are still running into issues.

Chris

Licheng-Guo, 2020-12-04:
I see, thanks!

BTW, I think it will be useful if there is an API to directly place the module where it is originally implemented by Vivado. Currently seems that I need to first get the anchor location, do some string editing to extract the site name, then place the ModuleInst at that site.

Thanks for your help!

closed 2020-12-04 by Licheng-Guo