Xilinx/RapidWright #97: Correct Ways to Remove Cell from Post PnR DCPs

Licheng-Guo, 2020-11-13:
Hi Chris,

I'm trying to remove some parts from a routed DCP. It seems to be working, but I'm not sure if I missed anything.

Specifically, I try to remove an FDRE called "snk", which is contained in a ModuleInst "p1"
```java
    // remove from the physical netlist
    design.removeCell("p1/snk");

    // remove from the logical netlist
    top.getCellInst("p1").getCellType().removeCellInst("snk");
    top.getCellInst("p1").getCellType().getNet("src").removePortInst("snk/D");
    
    // update the logical netlist
    netlist.resetParentNetMap();
```

I wonder if my understanding and operation about the two netlists are correct? At least it looks correct if I open the processed DCP in vivado.

I think in general the relationship between the physical netlist and the logical netlist is to some extent confusing. It will be great if you can add more explanation or examples to the tutorial! 

Thanks a lot!

Best,
Licheng

clavin-xlnx, 2020-11-13:
Hi Licheng,

Modifying the netlist inside of a ModuleInst is going to be a bit more work as these objects are backed by the Module from which it was instantiated.  To be consistent, you would have to also modify the Module, all derived ModuleInsts in the exact same way.  One way to avoid this mess is to simply flatten the design after you have created all the desired ModuleInsts.  This breaks the ties to the Module/ModuleInst objects so that you can modify the design freely.  You can do this for the logical netlist and physical netlist by calling:

```
EDIFTools.flattenNetlist(Design design); // Logical netlist flattening
design.flattenDesign() // Physical netlist flattening
```

To remove the cell and net, what you have looks reasonable.  Although, I don't know all the details of your design, I could not reproduce the error you mentioned above.  If you want to share your design with me (either posting here or sharing with me directly), I can take a closer look at the error.  

You are correct that there are several details about the relationship between the logical and physical netlist that must be maintained.  I will think more on how to best improve/communicate that information.  

-Chris

Licheng-Guo, 2020-11-18:
Hi Chris,

Thanks a lot for your answer! I'm testing it to see if things work out.

BTW, I think it will be perfect if an example can be provided to show how to do simple editing of the post-routed netlist and keep the physical and logical netlist synchronized. I think currently there is only an example to show how to edit the EDIFNetlist. I can try to provide one after I figure things out.

Thanks!
Licheng

closed 2021-04-13 by Licheng-Guo