Xilinx/RapidWright #104: NullPointerException on Node.IsTied

litghost, 2020-11-20:
The Node.IsTied method appears to result in a NullPointerException on some Node objects.  Test case is in #103.  Example invocation:
```
java com.xilinx.rapidwright.tests.IsNodeTiedTest xc7a35tcpg236-1
==============================================================================
==                    Check Node.IsTied: xc7a35tcpg236-1                    ==
==============================================================================
             Load Device:     0.879s     59.640MBs
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.device.Node.isTiedToVcc(Unknown Source)
        at com.xilinx.rapidwright.device.Node.isTied(Unknown Source)
        at com.xilinx.rapidwright.tests.IsNodeTiedTest.main(IsNodeTiedTest.java:53)

```

clavin-xlnx, 2020-11-23:
The underlying cause of this problem is Issue #105, however, I have a fix that will go into the next release to avoid the NPE.

clavin-xlnx, 2020-12-04:
This should be fixed in [2020.1.6](https://github.com/Xilinx/RapidWright/releases/tag/v2020.1.6-beta).  This also adds APIs to check if a BEL supplies GND/VDD: `BEL.isGndSource()` and `BEL.isVccSource()`.

closed 2020-12-08 by litghost