Xilinx/RapidWright #105: NullPointerException and/or incorrect Node's returned from Wire.getNode method

litghost, 2020-11-20:
The `Wire.getNode` method appears to return the incorrect `Node` or even cause a NullPointerException.  Test case is in #103.  Example invocation:
```
==============================================================================
==               Check Node -> Wires -> Node: xc7a35tcpg236-1               ==
==============================================================================
             Load Device:     0.872s     59.640MBs
Exception in thread "main" java.lang.RuntimeException: Got different Node on Node -> Wire -> Node, node 1 = WR1BEG_S0 node 2 = CMT_TOP_WL1END3_12
        at com.xilinx.rapidwright.tests.NodeToWiresToNode.testNode(NodeToWiresToNode.java:41)
        at com.xilinx.rapidwright.tests.NodeToWiresToNode.main(NodeToWiresToNode.java:78)
```

The current code in #103 uses:
```
            if(!node.getWireName().equals(nodeFromWire.getWireName())) {
```
for equality checking, but:
```
            if(!node.equals(nodeFromWire)) {
```
fails too.

If the check is completely removed, the following error is produced instead:
```
java com.xilinx.rapidwright.tests.NodeToWiresToNode xc7a35tcpg236-1
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.device.Tile.a(Unknown Source)
        at com.xilinx.rapidwright.device.Node.a(Unknown Source)
        at com.xilinx.rapidwright.device.Node.<init>(Unknown Source)
        at com.xilinx.rapidwright.device.Node.<init>(Unknown Source)
        at com.xilinx.rapidwright.device.Wire.getNode(Unknown Source)
        at com.xilinx.rapidwright.tests.NodeToWiresToNode.testNode(NodeToWiresToNode.java:38)
        at com.xilinx.rapidwright.tests.NodeToWiresToNode.main(NodeToWiresToNode.java:79)
```

clavin-xlnx, 2020-11-23:
This is very similar to Issue #63.  I've looked into trying to resolve the missing Node references but getting all of them requires a non-trivial amount of effort on the internal RapidWright side.  

Reproducing this issue, is rather straight-forward with a "for all" loop.  However, in practical settings, it is unlikely to have much use for a Node reference from arbitrary Wire elements that do not have much significance.  However, I am open to be persuaded otherwise.

litghost, 2020-11-24:
> This is very similar to Issue #63. I've looked into trying to resolve the missing Node references but getting all of them requires a non-trivial amount of effort on the internal RapidWright side. [...]

I'd argue you should either make `Wire.getNode` always work, or remove it.  Semi-randomly causing a NPE is not an acceptable implementation.

In terms of fixes, I've prototyped a solution for this (wire -> node) that was intended for the device database node <-> wire relationship.  It is general enough to be used in RapidWright to implement this properly: https://github.com/SymbiFlow/prjxray/pull/1477

clavin-xlnx, 2020-12-01:
I agree, NPEs are not acceptable and need to be fixed.  Modifying the backend implementation is expensive and I am continuing to explore to identify a lower cost solution.

clavin-xlnx, 2020-12-04:
This should be fixed in [2020.1.6](https://github.com/Xilinx/RapidWright/releases/tag/v2020.1.6-beta).  There is a small runtime hit, especially for larger devices when querying uncommon wires, but a message will print on the first instance and will be cached thereafter.

closed 2020-12-08 by litghost