Xilinx/RapidWright #106: Error running DeviceResourcesExample

reillymck, 2020-12-03:
With the latest updates to the DeviceResources code, now when I run 
`java com.xilinx.rapidwright.interchange.DeviceResourcesExample xc7a100t`
it produces the following error output:
```
Device Resources Dump: xc7a100t
             Load Device:     0.341s     72.630MBs
           populateEnums:     0.061s      6.622MBs
               SiteTypes:     0.037s      0.963MBs
               TileTypes:     0.053s      2.041MBs
                   Tiles:     0.031s      8.004MBs
             Wires&Nodes:    26.980s   1780.962MBs
            Prims&Macros:     0.227s     51.960MBs
    Cell <-> BEL pin map:     6.492s      1.021MBs
                 Strings:     0.012s      0.000MBs
              Write File:     8.155s      0.003MBs
Exception in thread "main" java.lang.RuntimeException: Duplicate BEL pin entry 'WEBWE[0]'
        at com.xilinx.rapidwright.interchange.CellBelMapping$CellBelPinMapping.readPins(CellBelMapping.java:37)
        at com.xilinx.rapidwright.interchange.CellBelMapping$CellBelPinMapping.<init>(CellBelMapping.java:89)
        at com.xilinx.rapidwright.interchange.CellBelMapping.<init>(CellBelMapping.java:180)
        at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyCellBelPinMaps(DeviceResourcesVerifier.java:481)
        at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:305)
        at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
```
I do believe the last commit where this command is functional for me is 2c5ed2c878eb40e8f1db1f22b99c65ea34fd65e2 on Oct. 14th

litghost, 2020-12-03:
I believe this is a duplicate of #101?

litghost, 2020-12-03:
Also I fixed bugs in this in #102, so try that.

litghost, 2020-12-08:
This should be fixed now!

closed 2020-12-10 by reillymck