Xilinx/RapidWright #107: issues with ubuntu 20.04

yichungchen84, 2020-12-05:
1. gradle 6 compiling error due to 
Execution failed for task ':compileJava'.
> directory not found: $RapidWright_root/bin/generated/sources/annotationProcessor/java/main
=>so I use "make", it works
2. Libpng12 -> need to be installed from PPA for ubuntu 20.04 (and 18?)

clavin-xlnx, 2020-12-07:
Thanks for pointing out these issues.  One solution that has been considered for Gradle is to use the gradle wrapper, this will lock down the version to be used with a known working set of commands that will compile.  I'll look into including that in a future release.

As for the Libpng12 need, this is due to RapidWright's dependence on an older Qt library for Java called Qt Jambi.  Eventually, we will move away from it, but for now it is a known issue (see [documentation]( http://www.rapidwright.io/docs/Automatic_Install.html)).

clavin-xlnx, 2021-05-19:
The gradle wrapper is now included with RapidWright as of 2020.2.5.  Closing this issue for now.

closed 2021-05-19 by clavin-xlnx