Xilinx/RapidWright #110: Likely integer overflow when using DeviceResourceExample on xcvu19pfsvb3824-1

litghost, 2020-12-08:
Error:
```
==============================================================================
==                 Device Resources Dump: xcvu19pfsvb3824-1                 ==
==============================================================================
             Load Device:     2.877s    553.966MBs
           populateEnums:     1.401s    145.259MBs
               SiteTypes:     0.172s      2.858MBs
               TileTypes:     0.575s      6.059MBs
                   Tiles:     0.990s    112.004MBs
INFO: Building uncommon Wire->Node cache...
      This might take a few seconds for large devices on the first call.  
      It is generally triggered when getting the Node from an uncommon Wire object.  
      To avoid printing this message, set Device.QUIET_MESSAGE=true or set the ENVIRONMENT variable RW_QUIET_MESSAGE=1.
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index -12546 out of bounds for length 55037
        at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
        at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
        at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
        at java.base/java.util.Objects.checkIndex(Objects.java:372)
        at java.base/java.util.ArrayList.get(ArrayList.java:459)
        at com.xilinx.rapidwright.device.d.a(Unknown Source)
        at com.xilinx.rapidwright.device.d.a(Unknown Source)
        at com.xilinx.rapidwright.device.d.b(Unknown Source)
        at com.xilinx.rapidwright.device.Device.d(Unknown Source)
        at com.xilinx.rapidwright.device.Device.a(Unknown Source)
        at com.xilinx.rapidwright.device.Node.b(Unknown Source)
        at com.xilinx.rapidwright.device.Node.a(Unknown Source)
        at com.xilinx.rapidwright.device.Node.<init>(Unknown Source)
        at com.xilinx.rapidwright.device.Node.getNode(Unknown Source)
        at com.xilinx.rapidwright.device.Wire.getNode(Unknown Source)
        at com.xilinx.rapidwright.interchange.DeviceResourcesWriter.writeAllWiresAndNodesToBuilder(DeviceResourcesWriter.java:528)
        at com.xilinx.rapidwright.interchange.DeviceResourcesWriter.writeDeviceResourcesFile(DeviceResourcesWriter.java:15
```

clavin-xlnx, 2020-12-08:
Thanks, I had not tested this device as its currently too big for exporting given our current implementation approach.  I was able to reproduce and provided a fix (bit field was an overflow).  The fix will be in the next release.

clavin-xlnx, 2020-12-11:
This should be fixed in 2020.1.7.

litghost, 2021-01-05:
When I tried this today (with 38e7dad247847ab422ac74cb1b6001bd12586980), I get the following error:
```
scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.DeviceResourcesExample xcvu19pfsvb3824-1
==============================================================================
==                 Device Resources Dump: xcvu19pfsvb3824-1                 ==
==============================================================================
Exception in thread "main" java.lang.ExceptionInInitializerError
        at com.xilinx.rapidwright.device.Device.getDevice(Unknown Source)
        at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:26)
Caused by: java.lang.IllegalArgumentException: No enum constant com.xilinx.rapidwright.device.Series.600
        at java.base/java.lang.Enum.valueOf(Enum.java:240)
        at com.xilinx.rapidwright.device.Series.valueOf(Series.java:31)
        at com.xilinx.rapidwright.device.PartNameTools.<clinit>(PartNameTools.java:63)
        ... 2 more
```

Replication instructions:
```
git clone https://github.com/Xilinx/RapidWright.git -b interchange
cd RapidWright
make update_jars
RAPIDWRIGHT_PATH=$(pwd) scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.DeviceResourcesExample xcvu19pfsvb3824-1
```

clavin-xlnx, 2021-01-05:
I believe you already identified the cause of this issue--I had forgotten to merge master into the interchange branch, which I have just merged your pull request.

litghost, 2021-01-05:
Indeed.  My test is getting further again now, and it's not quickly exploding.  I'll close this for now.

closed 2021-01-05 by litghost