Xilinx/RapidWright #119: Usability: Check matching versions

jakobwenzel, 2021-01-12:
Hi!

It would be great if instead of this:

```
Exception in thread "main" java.lang.ExceptionInInitializerError
	at com.xilinx.rapidwright.design.Design.setPartName(Unknown Source)
	at com.xilinx.rapidwright.design.Design.<init>(Unknown Source)
	at [...]
Caused by: java.lang.ArrayIndexOutOfBoundsException: Index 10 out of bounds for length 10
	at com.xilinx.rapidwright.device.PartNameTools.<clinit>(PartNameTools.java:57)
	... 7 more
```
we could get a message saying that the user forgot to update the data files.

Closed Source Code, Open Source Code and data files would each have to contain some kind of Version String. On start, the three versions could then be automatically checked for equivalence.

Kind Regards,
Jakob

clavin-xlnx, 2021-01-12:
Thanks Jakob, this is a good point.  We do version the device files and an error message is generated if there is a mismatch on load.  However, we do have other files in the `data` directory that do not have versions.  I can add those in the next release where there is a change to their content.

Chris

clavin-xlnx, 2021-06-23:
Starting in release 2020.2.6, all data files versioned and checked.

closed 2021-06-23 by clavin-xlnx