Xilinx/RapidWright #127: Duplicate pips from Tile.getPIPs()?

litghost, 2021-01-27:
Replication instructions:
```
git clone -b duplicate_pips git@github.com:litghost/RapidWright.git
cd RapidWright
make update_jars
RAPIDWRIGHT_PATH=$(pwd) ./scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.DeviceResourcesExample xc7a35tcpg236-1
```

Results in:
```
==============================================================================
==                  Device Resources Dump: xc7a35tcpg236-1                  ==      
==============================================================================
             Load Device:     1.026s     60.164MBs                                                       
           populateEnums:     0.187s      4.692MBs                                                       
               SiteTypes:     0.192s      0.963MBs
Exception in thread "main" java.lang.RuntimeException: Duplicate pip!!!!
```

clavin-xlnx, 2021-02-02:
Thank you, I have successfully reproduced the issue.  It appears that bi-directional PIPs are appearing twice in the results of calls to Tile.getPIPs().  I will work on providing a fix.

clavin-xlnx, 2021-02-03:
An updated jar has been made with a fix for the duplicate PIP issue:
https://github.com/Xilinx/RapidWright/releases/download/v2020.2.1-beta/rapidwright-api-lib-2020.2.1_update1.jar

This will be added to the next release (2020.2.2).

litghost, 2021-03-11:
@clavin-xlnx Has the latest JAR's been updated so that this issue is fixed?  If so, this issue can be closed!

clavin-xlnx, 2021-03-11:
This fix is included in 2020.2.2.

closed 2021-03-11 by clavin-xlnx