Xilinx/RapidWright #134: NullPointerException running Interchange.PhysicalNetlistExample

reillymck, 2021-03-11:
I am trying to run the PhysicalNetlistExample, as described in [interchange/README.md.](https://github.com/Xilinx/RapidWright/blob/master/interchange/README.md) Everything works great until I try to run the last line in the example.

When I use either of the following commands, I get a java.lang.NullPointerException.

`java com.xilinx.rapidwright.interchange.PhysicalNetlistExample pblock0.dcp pblock0_interchange.dcp`
`java com.xilinx.rapidwright.interchange.PhysicalNetlistExample pblock0.dcp pblock0.edf pblock0_interchange.dcp`

Here is the stack trace for reference:
```
Exception in thread "main" java.lang.NullPointerException
        at com.xilinx.rapidwright.interchange.Enumerator.getKey(Enumerator.java:19)
        at com.xilinx.rapidwright.interchange.Enumerator.getIndex(Enumerator.java:30)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writeRouteBranch(PhysNetlistWriter.java:389)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writeRouteBranch(PhysNetlistWriter.java:427)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.populateRouting(PhysNetlistWriter.java:376)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writePhysNets(PhysNetlistWriter.java:285)
        at com.xilinx.rapidwright.interchange.PhysNetlistWriter.writePhysNetlist(PhysNetlistWriter.java:462)
        at com.xilinx.rapidwright.interchange.PhysicalNetlistExample.main(PhysicalNetlistExample.java:45)
```

clavin-xlnx, 2021-03-11:
This is because the DCP being loaded has routing stubs in the clock network.  I believe this DCP might be out-of-context and thus doesn't have a fully routed clock network.  It is using a half-PIP to denote where it wants the clock to go when the design gets fully routed.

The Interchange format doesn't yet support this notion of a half-PIP and so it errors out.  Here is the half-PIP:

`RCLK_CLEM_L_X5Y209/RCLK_CLEM_L.CLK_BUFCE_ROW_FSR_0_CLK_IN null`

The end wire is null which is causing the problem. @litghost - To appropriately address this issue, it would likely require us to change the Interchange schema for a [PhysPIP](https://github.com/SymbiFlow/fpga-interchange-schema/blob/main/interchange/PhysicalNetlist.capnp#L131) to allow wire1 to be null.  Do you have any thoughts on this approach?

litghost, 2021-03-11:
@clavin-xlnx I wonder if out of context flow boundries make more sense as a new object, rather than a PhysPip with only `wire0` set.  Either way, I agree we need to extend the schema.  @clavin-xlnx What do you think about writing up a little doc (to go under [schema](https://github.com/SymbiFlow/fpga-interchange-schema)/docs) about this?

litghost, 2021-03-11:
@gatecat Do you have an thoughts around out of context flow representation?  I believe you were working on something adjacent to this.  If I'm mistaken sorry.

gatecat, 2021-03-11:
I sketched something in https://github.com/YosysHQ/nextpnr/pull/490/files but other pressures at the time meant it wasn't completed. I think the API ideas are still relevant, although could be coupled with improved handling of top level pins in general.

litghost, 2021-03-11:
@clavin-xlnx When an out of context flow is represented in the physical netlist, are `port` `PhysCell` emitted at the half-pips?  This would be good to document as well.

clavin-xlnx, 2021-03-11:
@litghost The ideal representation instead of a PIP would be a node.  I was looking for that, but it looks like we didn't add that to the schema and this might be a good reason to.  This particular instance is an artifact of the Vivado clock router.  It runs during placement and leaves breadcrumb nodes at strategic decision points to allow the router to fill in the gaps later.  

I don't think there are any `port` `PhysCell` that are emitted as part of this as this is all within the same net.

litghost, 2021-03-11:
@clavin-xlnx Agreed.  Given that these nodes have no upstream or downstream edges, I believe implementing them as a half-pip is not right.  Instead, let's add a `stub_nodes` field to `PhysNet`.  The contents of `stub_nodes` is simply the list of nodes that are part of a net, but have no upstream or downstream edges.  I believe there also needs to be "fixed" bool in the list to indicate whether that node is required to be part of the net eventually, or not.

clavin-xlnx, 2022-07-13:
Fixed as part of #467.

closed 2022-07-13 by clavin-xlnx