Xilinx/RapidWright #159: Failure verifying UltraScale+ interchange output

gatecat, 2021-04-20:
Running:

```
./scripts/invoke_rapidwright.sh com.xilinx.rapidwright.interchange.DeviceResourcesExample xczu2egsfva625-1
```

fails with:

```
==============================================================================
==                 Device Resources Dump: xczu2egsfva625-1                  ==
==============================================================================
             Load Device:     0.758s    234.752MBs
           populateEnums:     0.102s     11.074MBs
               SiteTypes:     0.092s      2.963MBs
               TileTypes:     0.078s      6.303MBs
                   Tiles:     0.033s      0.004MBs
INFO: Building uncommon Wire->Node cache...
      This might take a few seconds for large devices on the first call.  
      It is generally triggered when getting the Node from an uncommon Wire object.  
      To avoid printing this message, set Device.QUIET_MESSAGE=true or set the ENVIRONMENT variable RW_QUIET_MESSAGE=1.
INFO: Finished building uncommon Wire->Node cache
             Wires&Nodes:    34.848s   2584.050MBs
            Prims&Macros:     0.330s     79.320MBs
    Cell <-> BEL pin map:    14.039s      1.555MBs
                Packages:     0.004s      0.036MBs
               Constants:    13.457s      0.013MBs
                 Strings:     0.043s      0.001MBs
              Write File:    12.731s      0.003MBs
Exception in thread "main" java.lang.IndexOutOfBoundsException: Index -1 out of bounds for length 167
	at java.base/jdk.internal.util.Preconditions.outOfBounds(Preconditions.java:64)
	at java.base/jdk.internal.util.Preconditions.outOfBoundsCheckIndex(Preconditions.java:70)
	at java.base/jdk.internal.util.Preconditions.checkIndex(Preconditions.java:248)
	at java.base/java.util.Objects.checkIndex(Objects.java:372)
	at java.base/java.util.ArrayList.get(ArrayList.java:458)
	at com.xilinx.rapidwright.interchange.LogNetlistReader.readEDIFCellInst(LogNetlistReader.java:202)
	at com.xilinx.rapidwright.interchange.LogNetlistReader.readEDIFCell(LogNetlistReader.java:132)
	at com.xilinx.rapidwright.interchange.LogNetlistReader.readLogNetlist(LogNetlistReader.java:302)
	at com.xilinx.rapidwright.interchange.DeviceResourcesVerifier.verifyDeviceResources(DeviceResourcesVerifier.java:454)
	at com.xilinx.rapidwright.interchange.DeviceResourcesExample.main(DeviceResourcesExample.java:35)
```

Some basic debugging (printing out the parent cell name) suggests the problem is related to the `HBM_ONE_STACK_INTF` although I haven't been able to get further than that.

gatecat, 2021-04-20:
Looking into the generator code, I _think_ what's happening is related to these two lines:

https://github.com/Xilinx/RapidWright/blob/4aecfe94f043b982ab41374b3c585559d59b7d3c/src/com/xilinx/rapidwright/interchange/DeviceResourcesWriter.java#L234-L235

The primitive library is for the specific Zynq UltraScale+ device, which correctly excludes the HBM primitives; but the macro library is for the whole UltraScale+ series and consequently includes the HBM macro. The end result is a reference to a non-existent cell in the macro definition.

clavin-xlnx, 2021-04-20:
Thanks @gatecat, I added a fix that should filter out macros that are unused/unsupported by the target device.

gatecat, 2021-04-20:
I can confirm the issue is fixed, and the generation and verification completes successfully.

closed 2021-04-20 by gatecat