Xilinx/RapidWright #161: ArrayIndexOutOfRange in Tile.getWireConnections

trharoldsen, 2021-04-23:
I'm following the following path out of a BRAM.  `BRAM_L_X6Y30/BRAM_FIFO18_DOADO4 -> BRAM_L_X6Y30/BRAM_LOGIC_OUTS_B13_1 -> BRAM_INT_INTERFACE_L_X6Y31/INT_INTERFACE_LOGIC_OUTS_L_B13`.  The first two connections are fine, but when I get to the third wire, RapidWright fails with an `ArrayIndexOutOfBoundsException` when I call `getWireConnections`.

I have some example code below.  The part is a `xc7s25csga324`.

```java
Wire w = device.getWire("BRAM_L_X6Y30/BRAM_FIFO18_DOADO4");
System.out.println(w.getTile().getWireConnections(w.getWireIndex()));

w = device.getWire("BRAM_L_X6Y30/BRAM_LOGIC_OUTS_B13_1");
System.out.println(w.getTile().getWireConnections(w.getWireIndex()));

w = device.getWire("BRAM_INT_INTERFACE_L_X6Y31/INT_INTERFACE_LOGIC_OUTS_L_B13");
System.out.println(w.getTile().getWireConnections(w.getWireIndex()));
```

The exception trace is

```
Exception in thread "main" java.lang.ArrayIndexOutOfBoundsException: Index -1 out of bounds for length 0
	at com.xilinx.rapidwright.device.d.a(Unknown Source)
	at com.xilinx.rapidwright.device.d.b(Unknown Source)
	at com.xilinx.rapidwright.device.Tile.getWireConnections(Unknown Source)
	at edu.isi.cift.Router.main(Router.java:180)
```

clavin-xlnx, 2021-04-23:
Looks like a typo error inside `getWireConnections()`, I will have it fixed in the next release.  This is due to the fact that the last wire (`BRAM_INT_INTERFACE_L_X6Y31/INT_INTERFACE_LOGIC_OUTS_L_B13`) is an uncommon one so it has some different handling under the covers.  Here is a work around for the problematic line in the meantime:

```
System.out.println(Node.getNode(w).getAllDownhillNodes());
```

trharoldsen, 2021-04-23:
Thanks Chris.

clavin-xlnx, 2021-04-29:
2020.2.4 should contain the fix for this issue.